hzeller/rpi-rgb-led-matrix #699: video viewer to stream file conversion on PC?

mbates14, 2018-10-03:
Is it possible to make and use the video-viewer script on a PC installation of linux to convert video files over to stream files? 

I would like to use my PC to just convert all my shorts over to str files for use with the Pi without using the pi itself. 

is this possible under the current configuration? Just curious.

hzeller, 2018-10-03:
Yes.
Just make sure to provide the same parameters that describe the LED layout, such as `--led-parallel` or `--led-chain`. 
Also, the converting machine needs to be little endian, but Intel PCs are, so no problem there.

You don't have to run as root, as the conversion functionality does not need to access any GPIO hardware (which wouldn't be available anyway in your PC)

mbates14, 2018-10-03:
Perfect, thats what I needed to know ;-)

closed 2018-10-03 by hzeller