hzeller/rpi-rgb-led-matrix #701: Interfacing with 80*40 Panels

hemanthkd, 2018-10-09:
I got the new release of Qiangli P4 80*40 panels, when tried interfacing, I faced a problem that rows from 16 to 20 as well as 36 to 40 are not turning on. Hope  you would help us

hemanthkd, 2018-10-09:
Image, which I have given as input is download.jpg
[image]
And the result I got is Result.jpeg
[image]

hzeller, 2018-10-09:
You are not providing enough info. How did you connect it and what are the commandline options you provide to the program?

hemanthkd, 2018-10-10:
We took total of  8 panels in two chains and arranged in U mapper in two chains with the Active-3. So in one chain we place 4 panels in U-shape so that first two will be normal and other 2 will be upside down. And the commands we followed are:
pi@raspberrypi:~/rpi-matrix-pixelpusher/matrix/utils $  sudo ./led-image-viewer   --led-rows=40 --led-cols=80 --led-pixel-mapper="U-mapper"  --led-chain=4  --led-parallel=2  --led-no-hardware-pulse download.jpg

hzeller, 2018-10-10:
with 40 height, you need the E address line configured on the active3 adapter. did you?

hemanthkd, 2018-10-10:
The panels which we got from Qiangli have only four address lines (A,B,C,D) and let us know how can we configure things on active-3 adapter

hemanthkd, 2018-10-10:
I am extremely sorry that just now I have found the E pin on the 8th pin and have configured accordingly by connecting E pin and 8th, 4rth pin and GND on the active adapter but the result is the exact inverse of the previous output and is as follows:
[image]
as follows

hemanthkd, 2018-10-11:
When I am not connecting the E pin, I am getting the inital output(Result.jpeg). 
[image]

And when I have connected the Epin to the 8th pin on the adapter I found this output(UpdatedResult.jp
[image]
g).
It is something like if we merge both the outputs then its a complete output. Please help me out in resolving this issue

hzeller, 2018-10-11:
It looks wierd as if the `E` pin is permanently high.
Can you double-check that you have your UART disabled, as the E-line is sharing a pin with that, and maybe that interferes.
Also double-check the soldering on the relevant pins of the active3 board.

hemanthkd, 2018-10-12:
I have tried disabling UART using "sudo systemctl disable hciuart" but still the same result is taking place, but when I tried E pin connecting to GND physically, then the other  part of the screen is turning on as mentioned in the above pictures

hemanthkd, 2018-10-12:
And I also checked the solderings on the adapter board, those are clear and perfect. i hope the main issue is only due the permanent state of GPIO 15. Please help me out

hzeller, 2018-10-12:
did you disable the uart by putting `enable_uart=0` in `/boot/config.txt` ? Otherwise the system might attempt to write console messages to it.

hemanthkd, 2018-10-12:
Yes it is already present.

hzeller, 2018-10-12:
so your gpio15 is stuck, you have to get it unstuck. There is something going on with your hardware, either the breakout board or the Pi. Can't help remotely here, you have to systematically figure out where the problem is: measure the output while toggling it, replace hardware...

hemanthkd, 2018-10-12:
Should i use a new pi

hzeller, 2018-10-12:
depending if you find that the GPIO pin has a hardware fault

hemanthkd, 2018-10-13:
can I use some other pins  instead of GPIO 15 as I was not using the 3rd chain in the adapter

hemanthkd, 2018-10-25:
As discussed I have changed the GPIO 15 to GPIO 21 and also manually routes the adapter board, but this time the default state of GPIO21 is not getting changed and I am sure that both 15 and 21 are working fine for a simple blinking led code

davemaster, 2018-10-25:
Can You please provide a picture of your panels and your RPi connection (good quality pictures) so we can figure it the solution to help You?

hemanthkd, 2018-10-26:
I hope there is an issue with the command to enable the E line, while all the rest of the things are working fine. Here are some pictures of my panels and RPI connection.Link:https://drive.google.com/drive/folders/1dqstTAMj7qeeqWiZ4l9rJiKoGUE81TAL?usp=sharing

davemaster, 2018-10-27:
> 
> 
> I hope there is an issue with the command to enable the E line, while all the rest of the things are working fine. Here are some pictures of my panels and RPI connection.Link:https://drive.google.com/drive/folders/1dqstTAMj7qeeqWiZ4l9rJiKoGUE81TAL?usp=sharing

According Your pictures, You may have a connection problem. Try this, always, the simple first then the complex; so, try rpi-rgb-led-matrix with only ONE panel; also, use  those cables link these [https://www.sparkfun.com/products/11026](url); without the HAT. Try this, FIRST. Pictures or videos of this try please.

hemanthkd, 2018-10-28:
The same issue has taken place even I have directly interfaced the single panel with pi directly
Link for the pictures:https://drive.google.com/drive/folders/1u0MlPX2Oi_dZstOJsroZ5XfqVWDu4K12?usp=sharing

hemanthkd, 2018-10-31:
When I have connected the single panel and when given the following command

hzeller, 2018-10-31:
Alright, can you try again with a fresh checkout ? Hopefully that should make the 40 row panel working with `--led-rows=40`.

closed 2018-10-31 by hzeller

hemanthkd, 2018-11-01:
Thank you so much for your full support. Finally, I got the perfect output
[image]

muharramapri, 2020-12-14:
> Thank you so much for your full support. Finally, I got the perfect output
> [image]

now i have problem like you before, can you help me? please
[image]

commit 0fc35c5f (2018-10-31): Make E-address work for panels between 32 and 64 rows.

Hopefully fixes #701.