hzeller/rpi-rgb-led-matrix #702: C# example for displaying Image and Video

NeerajShrivastava04, 2018-10-12:
Hi Hzeller,

Thank you so much for writing such a wonderful library.

I am truing to write a led-image-viewer type of application in C# where I will pass on the image/video in C# program and it will show up on LED matrix. 
For this I have made the changes in existing led-image-viewer and have make another file in lib directory led-image-processor.cc. I have broken up the existing led-image-viewer main method code in two parts first is to prepare the image and other one is to display the animation. I have made all the required code changes and have exposed these two functions through graphics.h file to shared object (librgbmatrix.so).

I am facing an issue while trying to display the image through C# program. whenever I try to invoke the same its giving me the  below error :

`pi@raspberrypi:~/repo/rpi-rgb-led-matrix-master/bindings/c#/examples $ sudo ./example_c_sharp.exe
opt.gpio_slowdown = 1
options.GpioSlowDown = 1
FYI: Couldn't set affinity: No such file or directory
/usr/bin/cli: symbol lookup error: /home/pi/repo/rpi-rgb-led-matrix-master/bindings/c#/examples/librgbmatrix.so: undefined symbol: _ZN6Magick16InitializeMagickEPKc`

When I try to find this missing symbol in librgbmatrix.so then its showing me three references.

`pi@raspberrypi:~/repo/rpi-rgb-led-matrix-master/bindings/c#/examples $ strings librgbmatrix.so | grep _ZN6Magick16InitializeMagickEPKc
_ZN6Magick16InitializeMagickEPKc
_ZN6Magick16InitializeMagickEPKc
_ZN6Magick16InitializeMagickEPKc
pi@raspberrypi:~/repo/rpi-rgb-led-matrix-master/bindings/c#/examples $`

Can you please guide me that whether is it possible to show the images/videos through c# interface or not. In case yes then can you help in this regard.

Would highly appreciate if you or anyone else in this forum can help me over here.

Thanks in advance for such a wonderful library and all your help.

NeerajShrivastava04, 2018-10-24:
I have investigated it further that binary created in C# does not have the libgraphicsmagick++ linked.
Has anyone has any idea that how we can link this library in C# binary or is their any other way to display image and video in C# ?

davemaster, 2018-10-25:
Greetings,

Have You mono IDE environment installed? version?

NeerajShrivastava04, 2018-10-25:
No I have not installed mono IDE environment, I just installed MONO and compile my programs through command line. In case if you suggest to install mono IDE and if that will solve my issue then I am more than happy to install it. can you please suggest the version which would be required.

NeerajShrivastava04, 2018-11-21:
No I have not got the solution yet.

techsavvy123, 2018-11-24:
@NeerajShrivastava04  can you please tell me how did you solve your rebooting issue in this project https://github.com/freetronics/DMD2/issues/20 ,as i'm badly stuck here.