hzeller/rpi-rgb-led-matrix #703: ghosting problem in 3 rows and 6 columns 64 by 64 P3 Led modules

hemantjha28, 2018-10-19:
Hi,

I am facing a very peculiar problem. It would be great of you can help me find out the root cause and the possible solution.

I have a led screen with 3 rows and 6 columns 64 by 64 P3 Led modules. I am using active adapter board as suggested by you.

The screen works absolutely fine if I use only 12 modules in 3 row by 4 column configuration. But the moment we arrange it in a 18 module 3 by 6 configuration the ghosting problem appears on the extreme end.

We have tried using official pi power supply but that is not working with 18 modules. Then I powered pi with with LM256 DC Dc buck convertor so that I can vary the voltage. This is where i found out the following issue.  If I supply pi3 b+ a input voltage of 5.18 volts the ghosting problem occurs while if i increase the input splly voltage to pi to around 6 volts the ghosting reduces drastically although there is still some ghosting. I didnt try increasing the voltage above 6 as it way burn by pi.

IMG 1 is the result with input voltage 5.18
IMG 2 is the result with input voltage 6
IMG3 shows you the connections.

It looks like even with active adapters the output of 5v diminishes with the increase in the number of modules in a row. Is there a way to address this problem.

The current consumed by pi3 b+ was around 0.5 amps.

I am using a 40 amps power supply but as I am showing simple text the current drawn from the power supply is not more than 3 amps.

I am using the active adapter broad provided by you.

Thanks a lot in advance.

Hemant
[image]
[image]
[image]

davemaster, 2018-10-19:
Greetings,

Try using --led-slowdown-gpio=2 as You're using a RPi 3.

Best regards

hemantjha28, 2018-10-20:
Thanks a lot David,

It worked. 

Is it for all RPi 3 we have to use --led-slowdown-gpio=2 irrespective of the screen size?

Regards

davemaster, 2018-10-21:
Maybe with a bigger screen, You can use --led-slowdown-gpio=1 with the raspberry pi 3.

Please, take a picture of your screen that worked. By the way, did You use python to get that output on you screen.

Thanks in advance

hemantjha28, 2018-10-23:
Yes I was using python. Sure I will send you the image in couple of days as the screen is at a different place.

hzeller, 2018-10-27:
(disregard the above mention in the last commit. I meant #708 not 703)