hzeller/rpi-rgb-led-matrix #704: More than 2 rows per channel

CamiloBravo, 2018-10-19:
Hi Hzeller, thank you for your great work.
I have a question:
What is the maximum size I can have on a screen? I know that by enabling the 3 channels I can have up to 6 rows, but could I have more than 2 rows per channel?

davemaster, 2018-10-19:
Greetings,

You can do that using the "U-mapper"

So the following chain
    [<][<][<][<] }- Raspbery Pi connector

is arranged in this U-shape (on its side)
    [<][<] }----- Raspberry Pi connector
    [>][>]

example from
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use
<http://hzeller github rpi-rgb-led-matrix>

Best regards

davemaster, 2018-10-19:
You can do that using the "U-mapper" 

So the following chain
    [<][<][<][<] }- Raspbery Pi connector

is arranged in this U-shape (on its side)
    [<][<] }----- Raspberry Pi connector
    [>][>]

example from [https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use](url)

Best regards

CamiloBravo, 2018-10-19:
> Greetings, You can do that using the "U-mapper" So the following chain [<][<][<][<] }- Raspbery Pi connector is arranged in this U-shape (on its side) [<][<] }----- Raspberry Pi connector [>][>] example from https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use <http://hzeller github rpi-rgb-led-matrix> Best regards El vie., 19 oct. 2018 a las 10:00, CamiloBravo (<notifications@github.com>) escribió:
> […](#)
> -- David Elias Flores Escalante TeleTracking SAC

Thanks for reply!
Yes, I use "U-mapper" for that reason I have maximum 6 rows using the 3 channels.
I want to know if is possible add more than 6 rows, 8 for example.

Best regards

davemaster, 2018-10-19:
Ok,

May be If you update this library to do this:

[<][<][<] }----- Raspberry Pi connector (Channel 1)
[>][>][>]
[<][<][<]
[<][<][<] }----- Raspberry Pi connector (Channel 2)
[>][>][>]
[<][<][<]
[<][<][<] }----- Raspberry Pi connector (Channel 3)
[>][>][>]
[<][<][<]

just try changing the rotation, for example, in channel 1, the three middle panels need to be rotated 180 degrees.

[<][<][<]**[>][>][>]**[<][<][<] }----- Raspberry Pi connector (Channel 1).

The other solution, would be a Raspberry Pi "4" with more GPIOs available.

CamiloBravo, 2018-10-23:
Thanks David for your help! I going to try it.

vsb2007, 2018-11-13:
> Thanks David for your help! I going to try it.
Do you have any results?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin