hzeller/rpi-rgb-led-matrix #711: Printing unicode text on LED Panel

pgbhat, 2018-10-29:
Please guide me how to display an Indic language characters on LED Panel - e.g., 0C85 to 0C8A.

hzeller, 2018-10-29:
It should be possible to use the [otf2bdf](https://github.com/jirutka/otf2bdf)-tool to convert any TrueType font to BDF (There might be an issue though if these are outside the 16-Bit range of the old unicode standard, AFAIK the BDF format does not allow full 32-bit unicode. But it looks like it should for for characters such as U+0C85).

You can simply install otf2bdf on your debian machine with `sudo apt-get install otf2bdf`.

Now, use the TrueType font that contains the characters and run otf2bdf; in the following example, we are reading the true-type font `path-to-ttf-file.ttf`, and converting it to a 28pt size bitmap BDF font, stored in `/tmp/output.bdf`
```
   otf2bdf -v -o /tmp/output.bdf -r 72 -p 28 path-to-ttf-file.ttf
```

You then can use the BDF font, using the regular font API provided by the LED matrix (you can try it first with the `examples-api-use/text-example` provided in the distribution).

hzeller, 2018-10-29:
The API itself [accepts text in UTF-8 encoding](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L83).

hzeller, 2018-11-08:
Did you get it to work ?

pgbhat, 2018-11-09:
Thank you so much, Henner. Your input helped me to find a solution. 

Also, it is very kind of you enquiring if it worked. 

 

Regards,

PG Bhat

hzeller, 2018-11-09:
Cool, I can close this issue then.

arahasya, 2020-08-14:
Hi Henner,
After a lot of research on this topic
The above method of font convert won't work for languages like Devnagiri, Hebrew, Urdu 
Basically languages with complex ligatures and substitutions. 
The only way is to add support for OpenType text shaping libraries which handle these tasks like Harfbuzz.
Or a way around is use Qt for render text but it will require dependency on Qt libraries.
Do you have any ideas on how to achieve this for this library?

hzeller, 2020-08-14:
Yes, unfortunately, the old BDF fonts (from the 80ies) can only deal with Plane 0 unicode (u0000-uFFFF). And while Devnagiri, Hebrew and Urdu are in that plane, the current simplistic implementation assumes that the language is a LTR language and that each glyph is on its own without diacritic modifiers, which doesn't work for many languages. RTL could probably be implemented fairly straight forward, but more complex text shaping can not really be achieved with the current approach.

I am happy to accept a pull request if you'd like to add better text rendering, but keep the dependencies to a minimium, so Qt would be a bit heavy, more something along FreeType and HarfBuzz. I'd say start out with a separate implementation, that isn't tied to the current simplistic [`Font`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L28) but rather something that is useful in your context. We then later can see how we can consolidate simple BDF rendering and fully featured text shaping.

As for implementation: In general, this library provides a fairly simple [`Canvas`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h#L32) interface, so your text rendering essentially just needs to be able to call `SetPixel()`. In fact, the current [`DrawText()`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L121) is implemented entirely independent of the LED matrix, its only assumption is that there is a Canvas to draw to.

arahasya, 2020-08-15:
So I have found a working example. But I'm not sure how to integrate this. Can you have a look?

https://github.com/tangrams/harfbuzz-example

arahasya, 2020-08-31:
Hi,
So I tried rendering with Qt. Text is getting rendered of all fonts of all languages but there are some problems. There is some offset in x,y positions in the actual rendered text on matrix. It is different for different fonts. The spacing between the letters is uneven. This is because in monochrome mode sub-pixel rendering is not getting done properly. I tried all other settings but getting no better result.
As Qt also used FreeType, harfbuzz they will have the same problem. Do you have any solution in mind to this problem?
Only bitmap fonts work properly for such low resolutions. 

Comparison :-
1. Qt Full colour on black window 

[image]

2. Qt monochrome on black window.

[image]

Notice the spacing and quality

arahasya, 2020-09-05:
@hzeller?

hzeller, 2020-09-05:
Yes, for low resolution you need to look into bitmap fonts. TrueType fonts are not really meant for that, but if you use them make sure that all hints are used (I suppose TrueType has similar low-resolution hints as Type1 fonts have).