hzeller/rpi-rgb-led-matrix #712: Random Slight Flicker in P3 Screen

hemantjha28, 2018-10-31:
Hi 

Greetings !!!

We are noticing a slight flicker in the screen [1 row by 5 column P3 Module] whenever we show images on the same. It is not ghosting, the entire screen seems to have a slight flicker. It is not regular and mostly happens with images.

I am using a 5V 40Amps power supply. We are using a Pi3 B+, Active adapter with run time parameters like gpio slowdown = 2 as suggested in the earlier post.

The screen is not drawing more than 3 amps current so I am not sure if it is a power supply problem.

Thanks Again.

Hemant

[VID_20181031_125009311_HDR.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/2533266/VID_20181031_125009311_HDR.zip)

hemantjha28, 2018-11-02:
Hi,
Gentle reminder to check if someone can help us understand the root cause behind the problem and possible solution.

Do I need to share more details about the system?

Thanks

hzeller, 2018-11-02:
Please ake sure to read the README, in particular the troubleshooting section first.
Your power supply is not a problem.

It mostly boils down to: the Pi is very slow, so you need to make sure that there is nothing else running on it.

If you do, there is some mitigation to make slow flicker almost invisible at the expense of refresh rate:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105

hemantjha28, 2018-11-02:
Thanks .. Apologies to bother you. Will check with the troubleshooting section.

hzeller, 2018-11-08:
Could you fix it ?

hemantjha28, 2018-11-09:
Hi, we are not able to address it completely, at times it appears but it is quite less as compared to the situation earlier. Your suggestions have definitely helped us zero in on the real problem. Thanks Again.

closed 2018-11-09 by hemantjha28