hzeller/rpi-rgb-led-matrix #714: display camera output on rpi-rgb-led-matrix with picamera

matzehuels, 2018-11-05:
I am struggling with a project that I am working on currently. Essentially, I want to capture an image using picamera (see https://github.com/waveform80/picamera) and then display it on an RGB LED-matrix. It seems, somehow both packages cant work together and I can not really figure out why and how to fix this issue.

When I run the minimal example below, I get an error saying "* failed to open vchiq instance"

```# Capture image
from rgbmatrix import RGBMatrix, RGBMatrixOptions

# Configuration for the matrix
options = RGBMatrixOptions()
options.led_rgb_sequence = 'RBG'
options.rows = 64
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'
matrix = RGBMatrix(options = options)

# Output to LED Matrix

import time
import picamera
import picamera.array
from PIL import Image

camera = picamera.PiCamera(0)
stream = picamera.array.PiRGBArray(camera)
camera.resolution = (100, 100)

camera.start_preview()
time.sleep(2)
camera.capture(stream, 'rgb')

image = Image.fromarray(stream.array)
image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
matrix.SetImage(image)

hzeller, 2018-11-05:
Does the camera need root access ?
By default, the RGBMatrix construction will drop privileges to nobody/nobody. If you want to stay root, make sure to set the change the runtime options to not drop root privileges.

In the [c++ header](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix.h#L401) you see the description of the Runtime struct. You have to set the right value in Python.

hzeller, 2018-11-07:
did you get it to work ?

closed 2018-11-08 by hzeller

hygeanne, 2018-12-09:
Hello, 

I am working on this project as well and I came across is exact same issue. I did try your suggestion, but it did not work for me.

This might be an issue that has not been well documented yet, but I've tried various things fix it, including:
https://raspberrypi.stackexchange.com/questions/19436/how-can-i-permanently-fix-dev-vchiq-permission-errors

I ran this exact code in my setup, and tested each line. It turns out, that the line "matrix = RGBMatrix(options = options)"  was the line that's the problem, but I have no idea why. 
(Obviously, without it the matrix functions wouldn't work so those were commented out as well. But the "matrix = RGBMatrix(options = options)" line alone should have still executed). 

I played all the possibilities of where I could put that line, But I'd either get the same error, or the program would get completely stuck.

Is it possible that the GPIO pin of the camera and rgb-matrix could be conflicting with each other, thus not allowing access to the camera?

hzeller, 2018-12-09:
So did you try the `--led-no-drop-privs` option as suggested above ?

hygeanne, 2018-12-09:
Thanks for your reply! Yes sir, I did. Unfortunately, with no luck.

hzeller, 2018-12-09:
so your camera probably uses some of the functionality the matrix uses as well. If it is the hardware pulse generator, try `--led-no-hardware-pulse`.

matzehuels, 2018-12-09:
Sorry for getting back to this with delay. I got it running by not dropping the Sudo rights as hzeller suggested. Also remember to actually run the process as sudo!

hygeanne, 2018-12-10:
Thanks! I realized I actually wasn't changing the right value all along. 

Would you have suggestions as to how you would modify it to capture and display continuously? I would just wrap it into a "while true:" statement, but I can only get it to update once. Is there better way to do this? Any hints would be super helpful.

Thanks!

hygeanne, 2018-12-10:
Hi! Turns out I completely misread hzeller's original response as manually entering into the command line, not actually setting the value in the code. Thanks again!

marcmerlin, 2021-06-03:
Well, how about that, that was my problem too @hzeller , can't believe I didn't find this bug.
In my case I had to add 
    ropt.drop_privileges = -1;
before
    RGBMatrix *rgbmatrix = rgb_matrix::CreateMatrixFromOptions(defaults, ropt);
this allows the picam to still work after the RGBMatrix is initialized.