hzeller/rpi-rgb-led-matrix #717: how to write code for 32*32 rgb led matrix and raspberrypi 3to print a message on display

Abhimanyuabhi, 2018-11-10:
Hi sir....Good afternoon
Hope you have a good day.......
Sir I am newly working on raspberry pi 3 and RGB led matrix.
Sir..I don't know how to write code for the raspberry pi 3 and rgb led matrix. Now i want to write  a simple code for the 32*32 RGB led matrix.
/*****************************************************************************************************************/
 Sir I executed some sample  programs which are present in https://github.com/hzeller/rpi-rgb-led-matrix but I am getting only single output for all sample code.
/************************************************************************************************************/
Now I want to write simple code to print some text on the RGB led display...

Thanks you so much..........

Abhimanyuabhi, 2018-11-10:
Can any one help me........
"How to write simple code for 32*32 RGB led matrix by using raspberry pi 3 "
Code:1.  C_language
          2. C++
          3. python
          4.C#

maddogs, 2018-11-10:
I'd start with running the Python scripts from the 

/rpi-rgb-led-matrix/bindings/python/samples/ directory

Read through the runtext.py script and make a few changes to see how things work.

Abhimanyuabhi, 2018-11-10:
Thank you so much for reply sir...
but what changes should i do sir....please can you guide me sir...

maddogs, 2018-11-10:
We can't really help you until you confirm where you are so far. Have you successfully followed the documentation and got the initial demos running on your display as shown here

https://github.com/hzeller/rpi-rgb-led-matrix#lets-do-it

If so, have you then followed the Python documentation here

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

If you have what errors or problems are you having?

Abhimanyuabhi, 2018-11-10:
Thank you so much sir......
I executed this python code.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/runtext.py

 But I got few errors......
Traceback (most recent call last):
  File "runtext.py", line 6, in <module>
    from samplebase import SampleBase
  File "/home/avcc/rpi-rgb-led-matrix-master (1)/rpi-rgb-led-matrix-master/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/avcc/rpi-rgb-led-matrix-master (1)/rpi-rgb-led-matrix-master/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core

Abhimanyuabhi, 2018-11-10:
Thank you so much your reply sir.......
Yes...I am executes some sample demo

./demo -t 10 -D 0
but the output is not  in the form of expected....
it giving some blinking lights  that's it

Thank you sir.....

maddogs, 2018-11-10:
Before you go any further you need to get the demos running.

When you day blinking lights, can you post a photo.

Abhimanyuabhi, 2018-11-12:
Hai sir... 
A very good morning...
Hope you have a good day....
I run sudo ./demo  runtext.ppm example but  I am getting message. But it not in what we expected it giving half message only.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin