hzeller/rpi-rgb-led-matrix #718: Adapter Boards

jtaylor54, 2018-11-12:
First thanks you Henner Zeller for all your work on this project.

I've searched and could not find any solid info. If it's there please point me in the right direction and apologies.

Does anyone know any sources for premade adapter boards that will work with this library in case you didn't want to source all the parts and/or were not comfortable at SMD soldering?

Also does anyone have any experience or seen any work to create a web server and scripts to allow for secure admin access to upload and control displayed content? Android and iOS would be cool too. 

I'd be interested in any front-end work anyone has done or seen to extend this project and make it more accessible and user friendly as I think it has huge potential and would have wider appeal.

Thanks
JT

hzeller, 2018-11-12:
I think there are some premade adapters from some Chinese websites that are fairly cheap and work with this library. Don't know a link, but people have found it.

The project is primarily a library for you to do the things you need (and as such attempts to be as user-friendly as possible with a good C++ API).

If you need a a web-frontend in your context and other things you can use this library as the backend and add your code on top. I usually use it for instance with the [FlaschenTaschen]( https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md) protocol when I need simple remote update of the panel.

DaveyRance, 2018-11-21:
Not sure how it works but this board is available.
https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

I can confirm that the one at link below works took about 4 days to get to the USA.
https://www.acmesystems.it/HAT-A3

jtaylor54, 2018-11-22:
Thanks for the info
I ordered 2 of the ElectroDragon ones last week so they should be here soon
Jim

 
 

https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

https://www.acmesystems.it/HAT-A3

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin