hzeller/rpi-rgb-led-matrix #725: Different wiring neccessary for Pi 1 B+ (rev. 1.2)?

shivan, 2018-11-18:
I have the Pi 1 B+ (rev 1.2)

I already used it 4 years ago. Now I tried the current version with the wiring as described here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md.
But I get two horizontal lines with about 10 pixels length at line 1 and line 9.
Using 32x16 display.

What's wrong? Are there some pins to be exchanged? How can I find out?
I saw in the old version for example Pin 5 was used for CLK, now it's 11.
Or won't it work anymore with pi 1 b+?

davemaster, 2018-11-19:
have You tried?

--led-gpio-mapping="regular-pi1" referenced in [https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L19](url). 

Best regards.

shivan, 2018-11-19:
@davemaster 👍 👍 👍 

Great... !!!!!!!!

It works !!!!

but this wasn't documented on "wiring.md" :)

closed 2018-11-19 by shivan