hzeller/rpi-rgb-led-matrix #727: RPi 3 A+ crash

SorX14, 2018-11-20:
I've been running this project successfully on a RPi Zero W1.1 for a while, and recently bought a new 3 A+ to improve performance.

I originally tried swapping SD card from the Zero to the A+ and the A+ never booted. I figured something was incompatible, so I created a new Raspbian Lite image.

I turned off the sound module, cloned from the repo, and ran `make` for the demo program.

When I try to run it, the A+ just stops responding, all I see is this:

```
pi@led2-pi:/srv/test $ sudo examples-api-use/demo -D0
Size: 32x32. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs
packet_write_wait: Connection to fe80::xx port 22: Broken pipe
➜  ~
```
Is this library compatible with the new 3 A+? Or have I missed something obvious?

hzeller, 2018-11-20:
I have not tested it yet with the Pi3 A+, so it might well be that it is not working.

The Raspberry Pi people change memory sizes or start addresses of GPIO mmap() positions, so this might well be. A hard crash is likely the outcome in these cases.

There is likely some slightly different way needed to detect it.
Can you tell me what the output is if you type
```
  cat /proc/cmdline
```
 ?

If you change [Line 202 in lib/gpio.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L202) to 
```
  static bool DetermineIsRaspberryPi2() {
     return true;
    // ...
 }
```
Does it work ?

mckenzielong, 2018-12-06:
Same problem here.  Just an fyi: that change fixes crashes, but still some glitches happening:

[image]

mckenzielong, 2018-12-17:
Would it be a reasonable solution to change the `DetermineIsRaspberryPi2` to use `/proc/cpuinfo` instead of `/proc/cmdline`? 

As far as I can tell, this shows _BCM2709_ or _BCM2708_ depending on the device.

hzeller, 2018-12-17:
I am happy to accept pull requests that more reliably detect the Pis.

jdimpson, 2019-01-14:
This fixed the problem for me, too. I'm just noticing @mckenzielong has a proper patch. I'll check it out some time this week.

electric1490, 2019-02-28:
Hey folks! Just circling back on this thread. Overall, I had the same EXACT issue OP was seeing. 

After changing Line 202 of gpio.cc (referenced above) and executing `clean make ; clean`, I was able to get `sudo examples-api-use/demo -D0` running successfully. However, three headscratchers:

1) Raspbian GUI locks up after ~7 minutes of running the demo. Even after the Raspbian GUI freezes, the LED board will continue running the demo successfully in perpetuity. 
2) Python code doesn't appear to run whatsoever (neither my own code nor `sudo python simple-square.py` runs at all. The code will successfully execute in the command line as if it's running (CPU usage ticks upwards as well) but the LED matrix doesn't output any text or image).
3) After changing the gpio.cc file as per @mckenzielong 's pull request, I still don't have luck properly running python code. 

Anything I'm missing? I figure I must be missing a step somewhere. Any insight on next steps to try and figure out a solve? Thanks so much for your help with this!

Thanks!

hzeller, 2019-02-28:
Never run a GUI on a Raspberry Pi.

maxammann, 2019-04-23:
Can confirm this issue and the workaround on the PI 3+

commit 7fdbf1ed (2019-07-06): Update Pi Model detection; allow degraded non-root operation.

o Update PI model detection. Now detects Raspberry Pi 4 (fixes #840)
  and Rasberry Pi 3 A+ (fixes #727)
o If we can't mmap /dev/mem, fall back to /dev/gpiomem. Unfortunately,
  this will not allow to read the 1Mhz timer nor allow for PWM to work
  properly (fixes #680)
closed 2019-07-06 by hzeller