hzeller/rpi-rgb-led-matrix #728: Reversing width and height limits

silodec-ignacio, 2018-11-20:
Hello there.

I have been using your "rpi-rgb-led-matrix" program on GitHub for my 3x4 LED panel.

So far the code you've written has been great but I have run into a problem.

The normal set up for a panel like this would be:

chains
->>>>>>
[A1] [B1] [C1]  V Parallels
[A2] [B2] [C2]  V
[A3] [B3] [C3]  V
[A4] [B4] [C4]  V

With the chains going across from left to right and the parallels going from top to bottom. 

But the problem is we are using a Raspberry pi that can only handle 3 parallels not 4.

So we are plugging the connectors and setting a Pixel-Mapper that would map the panels so it has 4 chains and 3 parallels:

Parallels
->>>>>>       
[A1] [B1] [C1]  V Chains
[A2] [B2] [C2]  V
[A3] [B3] [C3]  V
[A4] [B4] [C4]  V

by setting the Chains going down in panels (Eg. From A1 to A2 to A3 to A4, as shown in the little diagram above.) 

And the parallels going across (e.g. A1 to B1 to C1)

I have set the Pixel mapper to change the X and Y coordinates to fit this set up but the problem is 
that by telling the flags that we have --led-chains=4 and --led-parallels=3 when I try to give it an X or Y coordinate that goes beyond it's limit
the Pixel does not show up in the display panel.

Because the width and height has been flipped I need to find a way to trick the program into reversing the width and height limits.

I have tried:
```
class CasArrangementMapper : public PixelMapper {
public:
  CasArrangementMapper() : parallel_(2) {}
  
  virtual const char *GetName() const { return "Cas-mapper"; }
  
  virtual bool SetParameters(int chain, int parallel,
                             const char *parameter_string) {
    return true;
  }
  
  virtual bool GetSizeMapping(int matrix_width, int matrix_height,
                              int *visible_width, int *visible_height)
    const {
	*visible_width = matrix_height;
    *visible_height = matrix_width;
	
	return true;
    }
	
 
  virtual void MapVisibleToMatrix(int matrix_width, int matrix_height,
                                  int x, int y, 
                                  int *matrix_x, int *matrix_y) const {
    
	/*
	LED Panel is going to be set up like this:
	[A1] [B1] [C1]
	[A2] [B2] [C2]
	[A3] [B3] [C3]
	[A4] [B4] [C4]

	But we need to map the pixels so that it instead reads across like:
	
	[A1] [A2] [A3]
	[A4] [B1] [B2]
	[B3] [B4] [C1]
	[C2] [C3] [C4]	
	*/
		
	
	// A1 --> A1
	if(x >= 0 && x <= 31 && y >= 0 && y <= 15){
		x += 0;
		y += 0;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// B1 --> A2
	else if(x >= 32 && x <= 63 && y >= 0 && y <= 15){
		// chain_length = 4 && parallel = 3
		x -= 32;
		y += 16;
		
		
		
		*matrix_x = x;
		*matrix_y = y;				
	}
	
	// C1 --> A3
	else if(x >= 64 && x <= 95 && y >= 0 && y <= 15){
		// chain_length = 4 && parallel = 3
		x += -64;
		y += 32;
		*matrix_x = x;
		*matrix_y = y;		
	}
	
	// A2 --> A4
	else if(x >= 0 && x <= 31 && y >= 16 && y <= 31){
		// chain_length = 4 && parallel = 3
			x += 0;
			y += 32;
		
				
		*matrix_x = x;
		*matrix_y = y;		
	}
	
	// B2 --> B1
	else if(x >= 32 && x <= 63 && y >= 16 && y <= 31){
		// chain_length = 4 && parallel = 3
		x += 0;
		y += 16;		
		
			
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// C2 --> B2
	else if(x >= 64 && x <= 95 && y >= 16 && y <= 31){
		// chain_length = 4 && parallel = 3
		x += -32;
		y += 0;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// A3 --> B3
	else if(x >= 0 && x <= 31 && y >= 32 && y <= 47){
		// chain_length = 4 && parallel = 3
		x += 32;
		y += 0;
		
		
		*matrix_x = x;
		*matrix_y = y;
		
	}
	
	// B3 --> B4
	else if(x >= 32 && x <= 63 && y >= 32 && y <= 47){
		// chain_length = 4 && parallel = 3
		x += 0;
		y += 16;
		
		
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// C3 --> C1
	else if(x >= 64 && x <= 95 && y >= 32 && y <= 47){
		// chain_length = 4 && parallel = 3
		x += 0;
		y += -32;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// A4 -- C2
	else if(x >= 0 && x <= 31 && y >= 48 && 63){
		// chain_length = 4 && parallel = 3
		x += 64;
		y += -32;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// B4 --> C3
	else if(x >= 32 && x <= 63 && y >= 48 && y <= 63){
		// chain_length = 4 && parallel = 3
		x += 32;
		y += -16;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	// C4 --> C4
	else if(x >= 64 && x <= 95 && y >= 48 && y <= 63){
		// chain_length = 4 && parallel = 3
		x += 0;
		y += 0;
		*matrix_x = x;
		*matrix_y = y;
	}
	
	
}
private:
  int parallel_;

};```​

If you could please return to me with a solution I would really appreciate it.

silodec-ignacio, 2018-11-21:
I guess I forgot to mention it in the description. The reason we can't use rotation is because we'll be using this LED display panel outdoors so the panel needs to stand up right. The panel has little bits of plastic above each LED light so the daylight doesn't make it harder to see the display message.

I've also ran into a problem with defining the amount of rows in the flag command line. The panels we have are 32x16, but when I put --led-rows=16 nothing shows up in the display panel.

davemaster, 2018-11-23:
This is a good idea. In Theory, this can be accomplished:

We can turn this,

chain 1 ==> [>] [>] [>] [>] [>] [>] 
                    [<] [<] [<] [<] [<] [<] 
chain 2 ==> [>] [>] [>] [>] [>] [>] 
                    [<] [<] [<] [<] [<] [<]
chain 3 ==> [>] [>] [>] [>] [>] [>] 
                    [<] [<] [<] [<] [<] [<] 
into 

[image]

with 32x32 panels we can get 192 width, 192 height or more, with 12 panels / chain. I think it's possible put more panels/chain with RPi3 or RPi3 +

daveythacher, 2020-09-12:
Something like this:
new_x = (((x * 96) + y) / 128)
new_y = (((x * 96) + y) % 128)

This will scale 96x128 (col x row) to 128x96, by giving each pixel a unique address. You are doing word width conversion, but the memory is not fundamentally changing. Therefore you just need to remap the address space, since you keep A1 A2 A3 A4 B1 B2 B3 B4 C1 C2 C3 C4 order.

daveythacher, 2020-09-12:
I am guessing here, but this is pretty old issue.

Just wire them:
A1 A2 A3
A4 B1 B2
B3 B4 C1
C2 C3 C4
*Letter is chain number, while number is panel number.

Then:
*matrix_x = (((x * 96) + y) / 128)
*matrix_y = (((x * 96) + y) % 128)

We are just reconverting the address mapping. Since we keep order. Data width is changing so we can treat it like a block of memory. (It is memory.)

So we go from a wired notion of 96 row by 128 column:
A1 A2 A3 A4
B1 B2 B3 B4
C1 C2 C3 C4

To a notion of panel notion of 128 row by 96 column:
A1 A2 A3
A4 B1 B2
B3 B4 C1
C2 C3 C4

closed 2025-11-27 by marcmerlin