hzeller/rpi-rgb-led-matrix #730: Invalid row sequence in 32x16 module

VadymKotsiuba, 2018-11-26:
I have a 32x16 module.
I connect the module to Raspberry Pi Zero W using Adafruit Bonnet.
I use the program rpi-fb-matrix to display part of the screen on the LED module.
For the test I use the image "**32x16_RECT_TRUE**".
As a result, on the module I see such as on the images "**32x16_RECT_FALSE**" and "**REAL_PHOTO**".

If you look closely, we see that the rows 0..3 are swapped with the rows  4..7; and rows 8..11 are swapped with rows 12..15.

I launch the program with the following parameters:
`sudo ./rpi-fb-matrix  --led-rows=16  --led-multiplexing=4  matrix.cfg`

Other values for "--led-multiplexing" do not give results, only spoil the picture and everything gets worse. 

What it can be?

Sorry for my English. I think my English very bad. I hope you understand what I said.
[image]
[image]
[image]

hzeller, 2018-11-26:
Please first start testing with the rpi-rgb-led-matrix project (this one) and play with the `--led-multiplexing` options and possibly `--led-row-addr-type` options to get the latest from there.
Then try in the `rpi-fb-matrix` project (which is derived and uses this library, but since I haven't written it, I can't help you with that)

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828