hzeller/rpi-rgb-led-matrix #731: What kind board can be low CPU with Pi Zero W?

JokerCatz, 2018-11-30:
first , sorry for my poor English ...

I tested Pi Zero W and Adafruit RGB Matrix Bonnet , this library is worked very well , thanx dev team : )

but Pi Zero cpu usage is very high , even it not refresh frames , and Pi Zero only single core ... so I try to find some hardware so pi make cpu usage down

is this
https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
or this
https://www.adafruit.com/product/2345
can be done I say ? make cpu usage normally without refresh ? or I only way is change to use addressable led matrix , like ws2812b ?

hzeller, 2018-12-04:
The CPU usage will be the same when you want to control this LED matrix, as it has to be constantly refreshed, there is no way around that. Best is to get a Raspberry Pi 3 and use the `isoclcpus`-setting as mentioned in the README.
Usually, you don't have to care about the CPU use, as this is your dedicated computer to do exactly this.

If you want less CPU, you can only use devices, that do the refresh themselves, such as addressable LED strips, correct.

JokerCatz, 2018-12-04:
okay , I tried change Pi zero to Pi 3 and split LED controller to another thread , the LED panel is not delay(blink) anymore , and it looking good : )

ws2812 or another addressable LED is too expensive ...

anyway many thanx dev team : )

closed 2018-12-04 by JokerCatz