hzeller/rpi-rgb-led-matrix #734: RGB Matrix and Webserver for Webpage

dodo124, 2018-12-04:
Hello, 

I am very impressed by your work and I got all demos running on a Pi3 with a driver HAT ( a red one for three panels in parallel).

But in your FAQ you mention that using three panels is a lot of work for the raspi (when running videos).

So do you think the following will work?

I want to run a additional webserver on the Pi3 and I want to show a html file on the rgb matrix.

The informations for the website (html file) are comming from several webservices like google, wetherforcast etc. There are quite alot of projects out there but all of them are not using this kind of rgb matrices. 

If it is works at all, what do you think will be the best way to implement this?

Regards Dominik

hzeller, 2018-12-04:
Three panels is no big deal, you can control many more with a Pi. Problem with video is, that the CPU of the Pi is too slow to decode and scale it fast enough.

If you have a HTML renderer that can render things into a canvas, you can do what you intend to do. It might be simpler though to have something that decodes the data from some json source and then have some manual dedicated renderers for each data; with the low resolution, you want for instance have a crisp bitmapped font without antialiasing (see the text-examples how to do that), or properly scaled icons (e.g. for weather icons such as sun, cloud and rain).

Should be a fun weekend project.

dodo124, 2018-12-04:
Hi again, I am more talking about 16 panels (64x64 matrices) 😉 but probably it will work. The initial idea was to use Apache and mindori and use fix scaled fonts there. But your idea might really be smarter because the performance should be much better. I could use some static text and variables for the external data. Can you recommend a source of knowledge e.g. book, tutorial?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828