hzeller/rpi-rgb-led-matrix #737: Question: Can you drive different chains with different utilities

lonedevwolf, 2018-12-13:
for example would it be possible to drive one chain 0 with led-image-viewer and in the meantime drive chain 1 with something else for example text via stdin or running a .ppm image.

i know its a long shot but i would realy be great if it is possible.

fusionfoto, 2018-12-13:
Have you tried it? It might take some modification to the code, but it seems like it should work.

lonedevwolf, 2018-12-13:
i haven't and i'm willing to try something but don't know where to start, so a direction would be greatly appreciated

hzeller, 2018-12-24:
You can _not_ use multiple programs to run different parts of the panel as they all need access to the same GPIO outputs. This will give a mess on the hardware leve.

So you need one program that runs the LED matrix. There, you have a large canvas, which you then can use by multiple parts of your program, even multiple threads if you want.
Or, you make it a networked display (e.g. with the flaschen-taschen protocol), and then use it as compositor for multiple independent programs.

closed 2018-12-24 by hzeller