hzeller/rpi-rgb-led-matrix #739: led-image-viewer "sticks" with jpg

GSheltonIII, 2018-12-18:
So I am using the led-image-viewer in a python script to view gifs with no problem. As soon as they are done with the loop, they stop (and play the next one as needed.) 

I have a jpg (ppm is the same result too) I want to put in the rotation. The issue is that it never "ends" the cycle. The command I am using is:

sudo ./utils/led-image-viewer -w5 -C test.jpg --led-rows=32 --led-cols=64 --led-chain=4

I have tried it with and without the -w5 and tried different times. (1 second, 10 seconds, etc..)

I have to ctrl C to stop the display of the jpg.

I figured out that if I do 2 images (even 2 of the same) then it "ends" as I expect. ie 
sudo ./utils/led-image-viewer -w5 -C test.jpg test.jpg --led-rows=32 --led-cols=64 --led-chain=4

So what am I doing wrong here? All help is appreciated. This library has been so awesome so far. It has given me a reason to begin to learn python.

hzeller, 2018-12-18:
you only provided one image, so there is no "cycle" to stop. The workaround is do is to just provide the same image twice as you found out.

GSheltonIII, 2018-12-18:
Perfect. That works for me. Thank you for the reply and all the work you have put into this!

closed 2018-12-18 by GSheltonIII