hzeller/rpi-rgb-led-matrix #745: Potentially new type of LED panel - no output

manuelgorman, 2019-01-05:
Hi,

I bought a couple of LED panels from ebay ([these](https://www.ebay.co.uk/itm/P3-RGB-pixel-panel-HD-video-display-64x32-led-Screen-module-dot-matrix-SMD-1pc/272157106020)). 

When I connect them to my Adafruit LED matrix bonnet, they don't do anything - if I disconnect them while leaving the power on, and reconnect, I get some random junk that resembles one of the demo's, but nothing usable (first image).

The chips are the following:
3x 74HC245C - right next to the data in connector
4x TC7258EN - in the middle, next to the power connector
24x FM6126A - along the edges in banks of 6

Image gallery [here](https://imgur.com/a/crVxT7K)

Datasheets:
[TC7258EN](https://datasheet.lcsc.com/szlcsc/Shenzhen-Fuman-Elec-TC7258E_C126213.pdf) - In Chinese but I can't find an English version, hope its of some use anyway
[74HC245C ](https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf)
[FM6126A](http://www.superchip.cn/Private/ProductFiles/FM6126A%EF%BC%8816Pin%E7%BA%A2%E5%A4%96%E9%81%A5%E6%8E%A7IC%EF%BC%89.pdf) - Also in Chinese

If I can provide any more info, please let me know

manuelgorman, 2019-01-05:
Thanks @electric1490, looks like a very similar issue but I'm afraid that thread looks like its on Arduino - I'm using a Raspberry Pi and an Adafruit Pi Bonnet. Not sure how much they have in common apart from the symptoms!

infurio, 2019-01-10:
Hi @manuelgorman, i think the symptoms look similar to what i was experiencing. Foolishly, i skim read the instructions and missed the section about disabling 1-wire first. Many hours later, disabled it and it just worked. Check out the gallery and see if you had the same issue.

[https://imgur.com/a/Q6ysiQ2](url)

shades66, 2019-01-13:
Hi,  saw your comment on issue #746  ,  i just added some comments to that with how I got these panels working with the adafruit matrix hat.,

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828