hzeller/rpi-rgb-led-matrix #747: LED Matrix Clears and Redraws Text on Update

electric1490, 2019-01-05:
First off, love this project. Super helpful and humbling as a first-timer. I've been able to get everything running as needed but I've run into one issue that I can't seem to get around.

- For some background: I use the LED panel to display countdown times. With that, I ask the script to refresh the panel every 10 seconds in order to display the most updated countdown (all generated in text via python).
- Issue: when the script loops to pulldown the latest values and turn around to display them, the screen goes completely blank for about a half second, and THEN generates the proper values.
- Video Showing Issue: [https://youtu.be/oHgf6tMPYYM](https://youtu.be/oHgf6tMPYYM)

Is there any way to eliminate this "refresh" blink when regenerating text output? Any help or insight would be hugely appreciated! Thanks!

hzeller, 2019-01-05:
(Can't see the video)
I suspect that the generation of the text image takes some time, so that is why you see the validation.

So you probably want to generate the text in an offscreen canvas while the countdown is displaying, and then use the `SwapOnVSync()` to swap it out.

In the clock, you find a C++-example of these concepts [`examples-api-use/clock.cc`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/clock.cc) (relevant part between line 150 and 183), you might be able to convert these ideas to Python.

electric1490, 2019-01-05:
First off - huge thank you for making this project and the lightning quick response. I wish my paid coworker were this fast :). 

I had a hunch that I'd have to leverage `SwapOnVSync()`. I was digging thru the samples to troubleshoot and thought that it looked like a viable path forward. 

I fixed the video from Private to Unlisted so you should be good to see it now. As long as you still think `SwapOnVSync() `is the best way forward, it looks like [`examples-api-use/pulsing-colors.py`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/ef550e49b91a36fae616ba8879e467e04da9552f/bindings/python/samples/pulsing-colors.py) has some python functionality I can learn off of as well. Thoughts on that? 

Again, huge thank you.

hzeller, 2019-01-05:
ah, thanks, can see the video now.
Yes, the pulsing-colors.py looks like it can provide a good starting point for you.

Essentially, you draw things in the offscreen canvas, then atomically swap it with the currently displayed one. It will probably only add a few lines of code to your current project.

electric1490, 2019-01-05:
Perfect, pretty sure it'll do the trick now that I look at it a bit closer. Let me noodle with it and let you know how I make out.

Thanks!

electric1490, 2019-01-05:
@hzeller , `SwapOnVSync()` ended up working like a charm. No blink whatsoever. Such a cool feature and great feeling to get one of biggest hurdles out of the way. 

Thanks again for the guidance in the proper direction. Saved me some therapy sessions :).

closed 2019-01-05 by hzeller