hzeller/rpi-rgb-led-matrix #750: Memory Problem

tuan91, 2019-01-10:
Hi, first off all, i want to thank you for the library. I have a Memory Problem. After 2 days my program is killed. My panel gets weather data from a REST API.  This should be done continuously.  Every 40 minutes, the RAM memory increases by 1%. I'll leave you an example code in this mail.. I hope you can help me, thank you! 
[image]

hzeller, 2019-01-10:
why are you creating a new frame canvas with every run_led_panel? you only need this once when creating the matrix. Otherwise you collect framecanvases in memory that you only use once.

tuan91, 2019-01-13:
Thanks for the quick feedback!
I thought I had to do it that way. I once tried to write "createframecanvas" outside the run_LED-panel, but it didnt work. Can you give me an example of what you mean, many thanks!

angelogoncalve, 2019-01-19:
Can you share the code how you fixed about the frame canvas?

Thank you for your help.

tuan91, 2019-01-21:
[image]

closed 2019-01-21 by tuan91

hzeller, 2019-01-21:
how can you call `matrix.CreateFrameCanvas()` when only in the next line you define `matrix = RGBMatrix(options = options)` ? Shouldn't these lines be swapped ?

tuan91, 2019-01-21:
Oh sorry, the two lines must be swapped.
Thank you for the Feedback!

angelogoncalve, 2019-01-21:
Dea Sir. Thank you for sharing the code and I will change in the code the two lines swapping them creating the matrix first and the canvas after.