hzeller/rpi-rgb-led-matrix #751: led-image-viewer

whatismyname185, 2019-01-13:
So, I did the sudo apt-get update
then I did sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y
but when I do the make led-image-viewer, i'm getting make: *** No rule to make target 'led-image-viewer'.  Stop.

I'm using a raspian stretch fresh nothing installed in it.

Can you please help me..

d3d9, 2019-01-14:
It seems like you didn't clone the repository (at least it isn't in your /home/pi directory), you need to ```git clone https://github.com/hzeller/rpi-rgb-led-matrix.git``` if you haven't already, then enter the rpi-rgb-led-matrix directory and run ```make all```.

whatismyname185, 2019-01-14:
OMG!!!...I got it to work...Thanks so much!!!, God Bless

closed 2019-01-14 by whatismyname185