hzeller/rpi-rgb-led-matrix #753: More than 11 bits for PWM?

kalan, 2019-01-20:
I am using three 64×32 LED panels in an extremely low-light environment (think of a bedroom with outside light completely blocked by curtains — and someone trying to sleep). There, I would like the panels to be really dim but still able to show several shades of gray.

In this setup, I fed a picture filled with RGB color #020202 into the demo 1 with default settings (brightness 100, in particular) — and realized that this is the perfect realization of white in my environment. However, neither trying to display #010101 nor going through all the brightness values with #020202 happened to display an intermediate shade — what I got was either the same color or complete black.

It seems to me that the only workaround would be to allow more than 11 bits for PWM. Grepping the code for the number 11 didn’t give me an answer why this number was chosen. Before I start hacking, I would like to ask: will increasing the maximum to, say, 15 require any significant changes to the code — and if yes, where to expect them? Or maybe there’s a simpler way to accomplish what I want?

depili, 2019-01-20:
At our hacklab we had a similar problem with some outdoor led modules that
we got our hands on. Adding more pwm levels was quite infeasible due to
the signaling speeds, but we came up with temporal dithering scheme which
adds some noise in the image but gives more apparent colors.

Our code wizard bypassed the 8bit rgb stuff altogether and in the end we
had a system with ~16 bits per channel and really wide dynamic range.
The code is in here: https://github.com/mankeli/hlsudp-rpi but in true
hacklab fashion the documentation is pretty much "read the source..."
sadly.

hzeller, 2019-01-20:
The value was choosen because much more would reduce the refresh rate so much that flickering will be much more noticeable as there is only a limited speed. Having said that, with the `--led-pwm-dither-bits` option, you might get a bit or so more out of it.

Brightness reduction I'd probably add by just adding more time between refreshes.
Another approach to get more color richness would be to calculate two images with intermediate values and swap them out on every refresh, doing some sort of dithering, probably similar to what @depili was doing.

kalan, 2019-01-20:
So I went on trying different values of `kBitPlanes` in `lib/framebuffer.cc`, as well as trying different values of `led_pwm_dither_bits`. (During the experiments, I had `FIXED_FRAME_MICROSECONDS=10000`, which I had chosen earlier to get rid of flicker caused by I/O.)

The best option turned out to be `pwm_bits=14` and `pwm_dither_bits=2`. With that, I am able to set `brightness=1` (of 100) and still have some 4 bits of color (at least, I could see #101010 but couldn’t see how it’s different from #181818; not sure if there’s some non-linear transformation in the way). This is definitely much more dimmer than anything I was able to achieve with `kBitPlanes=11`.

At `kBitPlanes=16`, I only got (somewhat expected) segfaults.

Having `kBitPlanes=15` made annoying flickering unavoidable, no matter what were `pwm_bits` and `pwm_dither_bits`. In particular, the setting of `pwm_bits=11` and `pwm_dither_bits=0` (which worked fine before the experiments — at `kBitPlanes=11`) resulted in flickering as well. Moreover, this setting gives flickering (and slightly brighter light) even at `kBitPlanes=14`.

So at least values up to 14 are not useless and should be available. 15 may be useful for certain setups too.

At first I thought of preparing a pull request with `kBitPlanes=15` (and all relevant places changed), but the last result makes me suspect that this could break many setups. Why do the same parameters work differently with different `kBitPlanes`? Is this effect unavoidable and thus `kBitPlanes` should be actually made a Makefile option? Am I missing something here?

Gansgar, 2019-09-07:
With the release of the Raspberry Pi 4 and it's new speeds of 1.5GHz instead of 1.4GHz, I suspect that the GPIO refresh rate has gotten even higher. Couldn't it be more possible now?

In theory my panels support a bit depth of 12. It would be awesome if one could use those.

hzeller, 2019-09-07:
the speed of the pi is not the problem, it is more a problem of the matrix speed. But for outdoor panels with high multiplexing rates it could possibly work, I am going to experiment with that.

Gansgar, 2019-09-07:
Hm okay. Than I've a question how the value of 12bit color depth on my panels come together. I've some 64x64, 1/32 multiplexed panels.
Wouldn't it be in theory be possible to drive them with 12 bit depth?

daveythacher, 2020-08-13:
Honestly the highest sane setting you can make would be 12 bits for a 8x16 RGB matrix. That would produce about 95 FPS. These use serial shift registers which have a max of 25MHz usually.

PWM shift register panels would work differently. However I am not sure this project supports them. With those there is usually a 25-33MHz grayscale clock. Here you become bound by pwm bit depth and multiplex rate against FPS. With shift registers you are bound by serial clock rate, pwm bit depth, multiplex rate and columns against FPS.

Note this project may not be capable of the full 25MHz. It may do less or burst. Because of this setting proper OE time is difficult. Hardware could help but Pi is limited here.

daveythacher, 2020-08-16:
Here is a link for what you can expect in terms of PWM bits at 25MHz depending on panel type/size. However this project likely does not work at 25MHz and does not support PWM or MM drivers.

https://github.com/hzeller/rpi-rgb-led-matrix/issues/918#issuecomment-674386191

daveythacher, 2020-08-16:
I think we can close this. More than likely this is related to the OE issue I created or a brightness scaling logic issue. It is clear that this issue asks for more PWM bits, which I do not think is reasonable for this project. There for I would view it as a different issue from say a brightness defect. However not enough information is provided to verify if such an issue exists.

hzeller, 2020-08-16:
@kalan I've now made it a bit easier to change the defaults, but you still have to recompile.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer-internal.h#L68-L84

In general, what you attempted with more bits and using dither bits should work.

There are newer panels, that actually allow to configure the current level (the FM612* kind of panels), which is a [todo](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L483-L484), but is not yet supported in this library, but might if I ever have such a panel to play with or if someone wants to supply a patch.

closed 2020-08-16 by hzeller

marcmerlin, 2020-11-10:
@kalan  @Gansgar  have you played with -led-pwm-dither-bits?
If so, can you chime in on https://github.com/hzeller/rpi-rgb-led-matrix/issues/1193 ?

daveythacher, 2020-11-11:
Please ignore marcmerlin and #1193. He is a contributor but actively spreads misinformation and has attempted to hijack discussion forms by forcing it into his forum.

pwm-dither-bits in what is a brightness issue is nonsense. The issue here is you either want to reduce the brightness or are having issues with full range of large PWM bit counts appearing. The current limit set by driver is a factor, multiplex ratio is factor, FPS kind of is factor, PWM bit count is a factor, BCM/PWM timing is a factor, etc.

Now to reduce the brightness and get multiplex steps will require increasing the PWM bit range. For example if I want to lower the brightness by 2 for a 8 bit image, I would use 9 bits of PWM. The MSB will always be zero, meaning for half the period it will be off. Thus the average current or brightness will be half. The issue is this will drive down the display frequency for serial shift register LED drivers. Assuming you configured it correctly, which is pretty tricky.

Misconfiguration will also cause the lower pwm bits to be dim or non existent. See #1120 for more information. Other LED drivers handle this a little more gracefully. However are not supported here. As a follow up to #1120 I created #1194. After having seen a number of users and "contributors" struggle with this.

Dithering is a quality reduction on average to to improve FPS. However if you misconfigure it will do nothing. (See #1193 for more information.) Configuration is very manual but allows for a number of flexibility in responding to certain usage scenarios. Overall it could have been done differently, however documentation was another option. 

Then again one really should understand what they are doing here assuming they actually care. As in not spreading misinformation, creating duplicate issues, trolling commenters, bogus notions of standards which have never existed in software collaboration or open source to my knowledge (Which is funny cause this area really gets random.), etc.

High PWM bit counts are used to convert log LED brightness to linear brightness. This requires a lot of performance which will exceed limits of panel or reduce quality of multiplexing. Raspberry Pi instability may increase depending on setup and environment.

Now as I mentioned earlier the average current step may become so low with high PWM bits and large/small multiplex ratios that you will have no benefit. Especially if attempting to dim the display. Therefore the image information just does not have enough current to appear in. 

This is why 33MHz is used for PWM drivers. Which is never explained. It has to do with 16 bit color against multiplex ratio assuming 20mA. However I can possibly violate that assumption.

daveythacher, 2020-11-11:
The reason why dithering works in misconfiguration is actually in #1120 and #1193. I think I may have semi touched on it in #1194.

The truth is you would not have to deal with it if you configured it properly.