hzeller/rpi-rgb-led-matrix #754: P3-64x64-32S

gioreva, 2019-01-26:
I have two sets of panels with the same label, P3-64x64-32S
One works, another one I see only colors.
I use "active-3" adapter on  the first chain.
I have tried all possible combinations, but I have not solved anything.

[image]

[image]

gioreva, 2019-01-29:
I read carefully, but also moving the Jumper in all the combination does not appear the image, not even a piece.
With a panel everything is fine, with each other appear only colors.
Any idea for the attempts I could make?

gioreva, 2019-01-30:
Supplier says:
"When you Do Smart setting, Please choose the IC ICN-2038S."
What does it mean ? What test can I do?

gioreva, 2019-01-30:
http://www.xlix.ru/datasheet/icn2038s.pdf

hzeller, 2019-01-30:
I don't have such a board to test (where did you get your board, do you have an aliexpress link ?), so if there is something different from the regular boards, the best way is to make it work with your board by reverse engineering and reading the datasheet, then send a pull request.

gioreva, 2019-01-31:
This is link of aliexpress panel:
https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.0.0.dc104c4dSXGYFM&orderId=507707052068521&productId=32959774837

I studied the integrated circuits datasheet:
DP5020B = (TLC5926, TLC5927 texas)?
FM6126
ICN2038
They are all identical and compatible.

The signal on the connector does not give you any indication to understand what to try to change ?
I have not a a working controller to do reverse engeenering, I can only make passive measurements.

gioreva, 2019-01-31:
sudo ./led-image-viewer --led-rows=64 --led-cols=64 --led-slowdown-gpio=3 --led-pwm-lsb-nanoseconds=2000 v.png

This is slow refresh image.
[image]

But image is random during connector insertion.
After it does not change, that I send white, black or lines.

RiddleSingh, 2019-02-03:
I have a P3-64x64-32S Matrix with A B C D E (8th Pin on HUB75 Connector) as shown below
[image]
[image]

I tried to run (500x32) runtext.ppm from the examples with the below configuration.

sudo examples-api-use/demo -D 1 examples-api-use/runtext.ppm  --led-chain=2 --led-multiplexing=0 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=10 --led-no-hardware-pulse 

I expected the text to be appeared in the middle 32 rows however, the result is different)  
Below is the result
[image]

I've also tried with different --led-multiplexing=0 (0 to 8), --led-chain= (1 to 4), --led-rows=(16, 32, 64)  --led-cols=(64, 128).

So far the best result was what I posted in the above gif. Can Someone kindly help me to understand the issue. 

Thank you

gioreva, 2019-02-03:
I saw that flaw when the jumpers in my active 3 were wrong.
I believe that my panels are not HUB75, but some other standard.
I'm amazed, that your panel, which apparently has my same connector, you see something.
What hardware do you use?

RiddleSingh, 2019-02-03:
My issue was already answered in one of @hzeller posts. 

I was connecting the wrong pin on RPI3, I had to connect to Pin24 on RPI3 for it to get working. 

Thanks a lot @hzeller.

gioreva, 2019-02-04:
@hzeller 
The Adafruit Hat (Rev-B)  with --led-gpio-mapping=adafruit-hat-pwm
Is it perfectly equal to the active 3 without parameter ?
Or should I check?
The panel of @RiddleSingh looks identical to mine.
Where do I find the mapping?

hzeller, 2019-02-04:
@gioreva With that parameter, it works for panels with A, B, C, D adresses and if you have done the PWM hardware mod as described in the README. If you have an E address line, you need to also manually wire up an E-line also described there, which @RiddleSingh did.

gioreva, 2019-02-08:
I ordered the Adafruit HAT, to see if it appeared as @RiddleSingh , but as before.
Random image at the insertion.
I have no made, no jumper, no cut wire.

sudo examples-api-use/demo -D 1 examples-api-use/runtext.ppm --led-chain=2 --led-multiplexing=0 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=10 --led-no-hardware-pulse

gioreva, 2019-02-21:
Solved using script #746

closed 2019-02-21 by gioreva