hzeller/rpi-rgb-led-matrix #757: No output (Adafruit HAT, 64x32)

tokun3, 2019-02-08:
I have a 64x32 matrix connected through the adafruit HAT and I can't get anything to appear on the screen. I have multiple displays because I was planning on chaining them but none of them seem to work (I am only trying them out one at a time right now). So it seems unlikely that the display is the problem. The power supply that is connected to the HAT is able to power the HAT and the Pi on its own, so that does not seem to be the problem either. This was only for testing - the rest of the time I have the Pi connected to its own supply. 

I am running Raspbian Lite and I have disabled audio and confirmed that one-wire protocol is not enabled. I have tried to test the demos and always include '--led-gpio-mapping=adafruit-hat' and I have tried both '--led-rows=32 --led-cols=64' and '--led-rows=32 --led-cols=32 --led-chain=2'. I have also tried various multiplexing options. 

Does anyone have any idea what the problem could be?

ltapilot, 2019-02-11:
I am successfully using the 6mm pitch 64x32 panels from Adafruit with the RPi 3B+ and Adafruit hat. I got 4 of these panels working on a chain by setting "--led-rows=32 --led-cols=32 --led-chain=8" I also had to specify the "U" mapper as I folded the panels back on themselves to make a 64x128 display. I am essentially treating the 64x32 panels as TWO 32x32 panels. I haven't set the jumper and disabled sound yet, so I do get a bit of flicker, but I'm confident I can get rid of that with the jumper and hardware PWM mod. Note that I'm using a 5V 4A supply for the Pi and hat and another TWO 5V 10A supplies for the panels. These panels eat current for lunch; that's how they can get so bright (and subsequently blister your retina). In some instances a moderately large, fast capacitor should be soldered right next to the connector on the panels to prevent fast switching transients from causing problems. I hope this helps.

patrickhirsh, 2019-02-16:
There's a chance that you have a HUB75E connector instead of a HUB75 on your panel. I ran into the same issue and didn't even notice because they look identical. If that's the case, take a look at shades66's script halfway down this thread: https://github.com/hzeller/rpi-rgb-led-matrix/issues/746

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828