hzeller/rpi-rgb-led-matrix #758: running rpi-rgb-led-matrix on picore

bduong06, 2019-02-09:
Hi Henner,

I just want to report that running your library on picore results in a perfect frame.  No flickering or artifacts.

I had previously tried with raspberry-stretch lite and the realtime version, but needed the --led-slowdown-gpio=3 in order to get rid of the random artifacts.  However, even with this setting I still got intermittent flickering.

I managed to install and setup a beta version of picore (10.x), and no everything is perfectly display without the need for the --led-slowdown-gpio setting.

I just thought you would like to know.

Thanks, for your library.

Bao.

hzeller, 2019-02-10:
Thanks, that sounds interesting; I'll have to try that.
`--led-slowdown-gpio` is usually needed because the gpio pins are _too fast_ for the LED matrix, so this is a bit worrisome that you don't need it as it would imply that the GPIOs might be slower with PiCore. Or maybe it just improves timing gaps ? Will have to test.

hzeller, 2019-02-10:
Do you have a link to the picore beta 10.x version image you are using @bduong06 ?

bduong06, 2019-02-10:
Sorry, I didn't include any details.  If you are using 3B+, you will need to use 10.x:
https://distro.ibiblio.org/tinycorelinux/10.x/armv7/test_relases/RPi/

However, if you have the others, you should use the 9.x version.
https://distro.ibiblio.org/tinycorelinux/9.x/armv6/releases/RPi/

If you can, use the 9.x version, as it will save you the trouble of having to build a lot of stuff from source.

Bao.

gioreva, 2019-02-19:
Goood, can you help me to install ?

tce-load –wi git
tce-load -wi make
tce-load -wi nano
tce-load -wi gcc
tce-load -wi libavcodec-dev  (Do not exist)
tce-load -wi libavformat-dev  (Do not exist)
tce-load -wi libswscale-dev  (Do not exist)

Missing other, it do not compile the library and WIRINGPI

gpulido, 2019-05-23:
@bduong06 I'm trying to compile the library on the 10.x arm7 version.
I'm in the same status as @gioreva It do not compile the library, would you mind provide some help to make it work on piCore 10 in a raspi 3b+?
Thank you

bduong06, 2019-05-30:
Sorry Gioreva and Gabriel,  my gmail has these messages filtered.  I just
stumbled on these today.

You need to build most of the libraries from scratch.  Some of them you can
get it from the previous version via wget.  However, you can get a version
from picoreplayer that's already all built for you.
You can download it from here -
https://www.picoreplayer.org/main_downloads.shtml.  You just have to
disable the player functionality.  You just need to disable the sound and
disable the webserver.  There is even a realtime kernel version available.
I have been using the realtime kernel version with no problems.  The nice
thing about picore is the os is loaded in ram so you never have to worry
about shutting it down.

picoreplayer will get you up and running as soon as possible.  However,
since this is fairly niche, I would suggest you be self sufficient.  If you
want to go the self sufficient route, let me know.  I will make sure I see
these messages.

Bao.

gpulido, 2019-05-30:
Hello,
Thank for the information.
I was capable of building it using a combination of libraries from 9.0 arm7 and a repository I found (I will update with the link asap) where all the libs where compiled.

Does the realtime kernell behaves better in this case?

Maybe we could think on put together the bare minimun to build & run the @hzeller library so it can be used as reference.

ca1ebd, 2019-12-11:
> Hi Henner,
> 
> I just want to report that running your library on picore results in a perfect frame. No flickering or artifacts.
> 
> I had previously tried with raspberry-stretch lite and the realtime version, but needed the --led-slowdown-gpio=3 in order to get rid of the random artifacts. However, even with this setting I still got intermittent flickering.
> 
> I managed to install and setup a beta version of picore (10.x), and now everything is perfectly display without the need for the --led-slowdown-gpio setting.
> 
> I just thought you would like to know.
> 
> Thanks, for your library.
> 
> Bao.

Hi @bduong06,
I'd like to get this set up on piCore, can you give a brief overview of the steps you took to get the library running on piCore? (I'm using the piCorePlayer RT image btw)

Thanks,
Caleb

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828