hzeller/rpi-rgb-led-matrix #759: more 3 channels

ebelouet, 2019-02-13:
HI,

Is it possible to add  more 3 channels to inscrease th hight level panel ?

Sincerely
Eric

jaygarcia, 2019-04-18:
@ebelouet I'm not a hardware expert, but from what I can tell with the Schematic, the number of connections seems to be a function of available pins on the Pi. There is also an issue of bandwidth. Pumping out that much data taxes one core ~100%.

Im working on a 20x(64x64) panel display using multiple Pi 3B+'s and piping the data via TCIP/IP via a network switch.

ebelouet, 2019-04-19:
HI,
Thank you for your answering and I can make board to try this

jaygarcia, 2019-04-19:
@ebelouet have you seen this thread? https://github.com/hzeller/rpi-rgb-led-matrix/issues/144

ebelouet, 2019-04-19:
now yes but where can i find schematic of it ?

closed 2025-11-27 by marcmerlin