hzeller/rpi-rgb-led-matrix #762: Changing Color and query REST API

tuan91, 2019-02-19:
Hello Henner,
I would like to light up the header in different colors and query and output the rest of Api below. The Api query works but why not the changing colors. I hope you can help me, thanks.

[image]
The code goes further but that is unimportant

tuan91, 2019-02-19:
Oh sorry i forgot to delete the  " createFrameCanvas" under the last else if (p) and the "SwapOnVSync(canvas)" under the last else if (ROM_sk).

hzeller, 2019-02-19:
You did not describe what you mean with 'does not change color'.

Note, you do  not want to create a new frame  canvas for every loop, as you will leak memory. Instead, you want to create the frame canvas once (e.g. right after you have created the matrix) and re-use that.

I am wondering what you are doing at the beginning of run_LED_Panel(): there you draw somcehing on some canvas, just to crate anew canvas and proceed with that ? This doesn't look like it does what you think it does.

tuan91, 2019-02-19:
Sorry if I didn't formulate it correctly. I intended to change the color of the headline. For this I have declared a Variabe. It should be incremented in the method. If the variable has the value <100 the headline should have the color white, if the variable should have the value between 100 and 200 the headline has the color red. If the variable is> = 200 then it should be reset to 0.

I had thought so, but that doesn't work. Do you have any idea how to solve this?

Under the heading, the REST API is queried by the weather station.

hzeller, 2019-02-19:
look at your code: you are writing the "weather station" headline to a canvas you then never use.

tuan91, 2019-02-19:
I know what you mean, that was just an example code. My real code looks different. I've taken that away what does not belong and that still does not work.

Of course, "canvas.Clear ()" is at the top of the method. In the example it is wrong.
[image]

tuan91, 2019-02-22:
solved!

closed 2019-02-22 by tuan91