hzeller/rpi-rgb-led-matrix #765: Problem with Chinese P5 192x64 (64x32, 3 cols, 2 rows)

uros-skopus, 2019-02-26:
Hi,

Thanks for both great project and community established!
I have the problem with Chinese P5 outdoor module 192x64 (64x32, 3 cols, 2 rows), 1:16, SM16207 LED, specification and image are attached.

[image]

[16207s-badLED-chipset.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/2903219/16207s-badLED-chipset.pdf)

It was connected with Raspberry Pi 3 B+ using GPIO pins and active hub. I have tested with both official:
https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
, and Chinese hub, but without success:

[image]

Command I am running is:
**sudo ./led-image-viewer --led-rows=32 --led-cols=64 --led-chain=3 --led-parallel=2 {image_name}**

I have also tried all multiplexing options but without success (--led-multiplexing=3 is something closest to what should be expected). It's not the problem of correct puzzling of IDC inputs (bottom-middle-top), but the image doesn't even have proper parts scattered, more like it is blurred. Just to mention 3:1 images are used for testing.

My first question is whether this lib is compatible with these Chinese SM16207 LED modules over HUBs?
Also, is it a problem to use these 3-channel HUBS (bottom-middle-top) to display just 2 chains?
Or is it something else?

Thanks in advance!

uros-skopus, 2019-02-26:
Adding some test images:
1) **sudo ./rotating-block-generator.py --led-rows=32 --led-cols=64 --led-chain=3 --led-parallel=2**

[image]

2) **sudo ./rotating-block-generator.py --led-rows=64 --led-cols=128 --led-chain=3 --led-parallel=2**

[image]

3) **sudo ./rotating-block-generator.py --led-rows=32 --led-cols=64 --led-chain=3 --led-parallel=2 --led-multiplexing=4 --led-slowdown-gpio=2**

[image]

uros-skopus, 2019-02-26:
Those were 32x32 modules, that was the problem.

closed 2019-02-26 by uros-skopus

SanddrP, 2023-03-28:
Hi.
I have a matrix 64*32 SM16207 LED, 
Please tell me the parameters of the meters you chose for the matrix, thank you