hzeller/rpi-rgb-led-matrix #768: Question re: powerbar

puffycholo, 2019-03-04:
Where is the powersupply connected on the powerbar? I will be using a good old computer power supply to power up my 12 panels. Any help?

hzeller, 2019-03-04:
on my power-bar, I have thick wires connected to the center, so that the distance to any panel is as short as possible.

puffycholo, 2019-03-04:
Those thick wires are the ones connected into the powersupply? Can i ask for a picture so i can have an idea on how did you do that. Thanks!

hzeller, 2019-03-04:
In the wiring documentation is a picture of one of the panels I did:
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power
You see the nuts in the middle of the power bars, they are for the cables that go to the power supply.

puffycholo, 2019-03-04:
Where are the bars connected to? I cant seem to find a connection going to
the powersupply.

hzeller, 2019-03-04:
The two nuts is where the power cables are connected to.

puffycholo, 2019-03-04:
Sorry for the dumb questions. Hehe where are the nuts connected to? Where
is there source of power?

hzeller, 2019-03-04:
The cable and power supply is not in the picture; it is just a standard 60A power supply and doubled-up Gauge 8 wire.

puffycholo, 2019-03-04:
The wire is connected to a nut on the bar?

hzeller, 2019-03-04:
Yes, using a crimped-on ring lug.

puffycholo, 2019-03-07:
Hello! May i ask what material did you use for the power-bar?

hzeller, 2019-03-07:
The link I mention above contains all the details.

closed 2025-11-27 by marcmerlin