hzeller/rpi-rgb-led-matrix #769: 64x64 Panels not working right (split image)

smcneece, 2019-03-06:
Hello all, thanks for any tips you can give me I have been trying all kinds of things I've read on here to get these to run any of the demos properly. I've tried tons of combinations of command lines. 

Ran the python script in one of the other threads, and still no luck.

The panels I bought are these: https://www.aliexpress.com/item/P2-5-Indoor-high-definition-led-display-Module-SMD-3-in-1-RGB-full-color-led/32816409052.html?spm=a2g0s.9042311.0.0.451f4c4dMznBY8

The CLI Command: _sudo ./demo -D 0 -t 12 --led-rows=64 --led-cols=64_

As mentioned I have tried many other command options, but willing to try more, this is by far the best image of them all with the simplest command.

What I get looks like: 

[image]

The back of the panel looks like: 
[image]

Greatly appreciate any help!

glindner, 2019-03-07:
I am having the same issue as well. I got my panel from Adafruit and it looks identical! I may have to get future once from Aliexpress as well and save 70%!

Anyways, like I said I've got the same thing happening and can't determine the reason why either.

glindner, 2019-03-07:
Looks like you are using the same RGB Matrix Bonnet that I am, and I have the solution (well, Adafruit does):

Look at Step 3 referring to 64x64 displays - https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices

Need to put a blob of solder for gpio pin 8 to map it to signal E as shown in the following picture - don't solder all 3 pads - just the middle and pin 8! -
[image]

Some matrixes need to use pin 16 for signal E, but I don't know anything about that.

smcneece, 2019-03-08:
Thanks for the reply! Do you also still have to jumper pins 4 & 18 per their article?

https://learn.adafruit.com/motion-controlled-matrix-bed-clock/circuit-diagram

glindner, 2019-03-08:
Yes, I do have the 4-18 jumper in place as well. 

Good luck!
George

BENETNATH, 2019-03-08:
Hello,
i have a quite similar issue with 64x64 panels from china + one active-3-like board
split image with the same command.
i've done the two bridges mentioned on https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3 (not clean but tested and ok), but that did not solve the issue.
Am i missing something?
pictures here : 
https://photos.app.goo.gl/NUk7ZwH2vrViFyTH6

from what i read on the panel, E is on pin 8, do i need additional tweaking? 
Thanks for your GREAT project.

BENETNATH, 2019-03-22:
Anybody has the same issue with this kind of panel? Can someone confirm that these panels should work ??

glindner, 2019-03-22:
Benetnath, I do have a suggestion. Your pictures actually show E - GND (left jumper) and P4 - GND (right jumper). The board you are using has the hole references in a different order compared to the document you reference (active-3). I would unsolder the left jumper and place the jumper from the top hole (P8) to the bottom hole (E). I believe you need the E-Line to go to Pin 8.

Give that a try and let us know if that helps!

BENETNATH, 2019-03-23:
That did the trick !!
Thanks a lot for your help

OllmGrb, 2019-09-19:
Hello, 

I have an led matrix display which i got off ali express, p2.5 to be exact.
I am supplying it with 5v and up to 16 amps of current, hence power shouldnt be an issue.
However, I get no response from the screen when supply is connected. voltages directly on the panel are ok, and have tried with 5 different screens, all with no luck. 
Are the displays supposed to give any indication when supply is connected?

Thanks

rahulpunk, 2020-02-26:
problem is a,b,c,d,e, 5 bit adress

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828