hzeller/rpi-rgb-led-matrix #776: Good power supply - with 2 USB ports?

shivan, 2019-03-18:
Hi,

I currently have some flickering additional wrong pixels when using my Anker 24W 2,4A Powersupply.

https://www.amazon.de/Anker-Ladegerät-PowerIQ-Technologie-Motorola/dp/B00WLI5E3M/

[image]

I thought this is great, because I only need one power supply to plug in the LED Panel 16x32 and the Raspberry PI 3 B. But it seems somehow, that's not enough.

I'm searching for hours now, and cannot find any usb DUAL power supply with more than the 2,4A per plug than this one. But I do  not want two power supplies. 
On my first try I have a USB power supply with 1A for the PI1 B and 3A special power supply for the display. With this one I nearly do not have any flickering bad pixels.

Maybe a 3300µF 25V Capacitor (low esr needed??) could solve that problem? Or do I always have to use 2 power supplies?

hzeller, 2019-03-18:
You can also try to start setting `--led-slowdown-gpio` to a higher value to see if this thing changes something.
A capacitor is always a good idea, in particular if you intened to run this on low-end power supplies, such as USB power supplies.

To easier find a power supply, just drop your requirement that it is USB. just look for a 5V 5A (or more) power supply and not worry about USB. That might make it easier to find as there are lot out there. Just connect that to the LED panel to it and the Pi via the GPIO pins, so you don't need any extra USB cable (e.g. the [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) has a way to directly connect 5V that can power the Pi)

shivan, 2019-03-18:
Hi hzeller,

thanks. I forgot that I already tried several --led-slowdown-gpio to 1, 2, 3... did not solve.
Also tried with pwm parameters...
played with led-pwm-bits, played with led-pwm-lsb-nanoseconds, slightly improved but not as good as I would like... and sometimes did not help anymore....

Ok, so maybe easiest try can be the capacitor... but good chance that it will solve my problem?
I just wonder why it works great on Pi1 with an older panel... (the anker usb power supply)
Is the pi 3 consuming too much power?

hzeller, 2019-03-18:
(The Pi3 is faster, so it might overwhelm the LED matrix, which is why slowdown sometimes helps)

It is hard to tell if the capacitor will help, it is really necessary to have a look at the power-smoothness with an oscilloscope to know that. But trying can't harm :)

Make sure to have the cables from the power supply to the panel as short as possible. 30cm is good, a meter or two is probably already too long. In particular USB cables often have very flimsy wires inside, so the resistance means quite a bit of voltage drop and also the increased inductance means that the power supply can't react quickly enough to power changes. These issues are somewhat mitigated with a low-ESR caapcitor directly at power connector to the panel.

shivan, 2019-03-21:
Hi,

took those:
Elko NCC KY 3300µF/25V 105°C

[image]

And solved the problem.

So solution is:
- use the Elko NCC KY 3300µF/25V 105°C directly next to the panel
- Parameter --led-slowdown-gpio=4

(without the elko one pixel keeps flickering, with elko NO pixel is flickering)

[image]
[image]

Only have to solder it...

I think this could be a good information on your chapter about power.
Direct solution for PI3B

closed 2019-04-02 by shivan