hzeller/rpi-rgb-led-matrix #778: led panel quangli q10 16x32 1/2 on icn2038s doesn't work

StarostinVlad, 2019-03-20:
Instead of displaying a test image the screen just flickers, tried to connect in different ways-to no avail also raises doubts HUB75 pins. What could be the problem? Do these panels fit at all?

vapiper, 2019-03-20:
What command line params did you use?

StarostinVlad, 2019-03-20:
> What command line params did you use?

--led-rows=16 --led-row-addr-type=1 and try --slowdown-gpio=1 and 2

vapiper, 2019-03-20:
Did you try `--led-row-addr-type=0` or `--led-row-addr-type=2`?

StarostinVlad, 2019-03-21:
I tried, but it still doesn't work. It's probably just the connection

StarostinVlad, 2019-03-21:
Dear hzeller, help me please))

vapiper, 2019-03-21:
1. You'd check the wiring.  See README for correct wiring and `--led-gpio-mapping` param.  I use Q10 32x16 1/4 matrix with ICN2038 chip. The pin CS isn't used, and it connected to GND inside the module. 
You also need to make sure that the matrix is not damaged.
It's better to use an active board to avoid flickers.

2. You'd  figure out  type of row addressing. Write test app that only one pixel lit at coords 0,0 with any colour, except black. Then try different values `--led-row-addr-type` param.
The goal of the step is only one pixel must be lit at any position.
I think, that next params will be correct `--led-rows=4 --led-cols=128 --led-row-addr-type=2`.
The matrix Q10 32х16 1/4 has a physical mapping 64х8, Q10 32x16 1/2 - 128x4.

StarostinVlad, 2019-03-22:
Thanks for the answer. Checking the matrix, everything works. The parameters did not help - the lines could not be less than 16. Reconnected everything to RPi and got this picture:
[image]
halfs different brightness and blink

StarostinVlad, 2019-03-22:
Валерий давайте по russian

vapiper, 2019-03-22:
Напиши в личку valerius77@ya.ru.

jannoke, 2024-06-11:
I have exactly the same panel. It seems to be this panel. There are config files on 4th tab. Use google translate to translate page to english: [https://p10.ru/product/svetodiodnyj-modul-x10-polnocvetnyj-smd-320x160-6000-qiangli/](url)

I'm running it with adafruit hat. I have physically checked that panel has ICM2038 chip. I could also get it to somewhat work on 5A-75B controller, but text is offset (it seems that middle of image is on the edge, so if I put a text there then on top and bottom of the panel is half and half of the text).

./demo -D 0 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 --led-rows=16 --led-cols=32

I have tried all the --led-row-addr-type and  --led-slowdown-gpio variants. Also set --led-rows and --led-cols.

I can see some movement, but as there are no videos of how the actual demos should look like it's hard to guess how off it is.

closed 2025-11-27 by marcmerlin