hzeller/rpi-rgb-led-matrix #779: Support needed for libgpiod/CONFIG_STRICT_DEVMEM systems

eklitzke, 2019-03-20:
I am using a bit of a non-standard setup on my rpi, I'm running Fedora 29 with Linux 4.20 instead of the Raspbian images that most people are using. On Fedora it is not possible to mmap /dev/mem, even as root:
```
$ grep ^CONFIG_STRICT_DEVMEM /boot/config-4.20.16-200.fc29.aarch64
CONFIG_STRICT_DEVMEM=y

$ sudo strace -ff examples-api-use/demo -D0
...
openat(AT_FDCWD, "/dev/mem", O_RDWR|O_SYNC) = 3
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_SHARED, 3, 0x20200000) = -1 EPERM (Operation not permitted)
close(3)                                = 0
dup(2)                                  = 3
fcntl(3, F_GETFL)                       = 0x2 (flags O_RDWR)
fstat(3, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
write(3, "mmap error: : Operation not perm"..., 38mmap error: : Operation not permitted
) = 38
close(3)                                = 0
write(2, "RPi1: MMapping from base 0x20000"..., 53RPi1: MMapping from base 0x20000000, offset 0x200000
) = 53
write(2, "Must run as root to be able to a"..., 77Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command
) = 77
exit_group(1)                           = ?
+++ exited with 1 +++
```

The "right" way to access the GPIO ports is to use [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/), uses some sysfs GPIO stuff added to the kernel in Linux 4.8. One nice thing about this is that you can access the GPIO pins this way as a non-privileged user.

I might look into adding ligbpiod support to rpi-rgb-led-matrix myself. I'm not super familiar with the library so I'm not sure how much work it is to port code.

hzeller, 2019-03-21:
Note, rpi-rgb-led-matrix requires to quickly write to GPIO pins, so unless libgpiod allows direct access to writing to the GPIO registers, setting a word of bits in parallel, or a very thin layer around it, it is likely too slow.

Also, the matrix library makes use of other hardware functionality and needs to set registers in the PWM hardware subsystem. Often these addtional hardware abstractions are neglected in abstraction libraries.

So if you want to adapt the library to use such abstraction layer, you need to make sure that the above conditions are met.

You're probably better off configuring your system to allow accessing `/dev/mem`

depili, 2019-03-21:
Quick glance into the libgpid source reveals that it just uses the
/sys/class/gpio character devices. So there would be the kernel
abstraction into the character devices and then the libgpiod abstraction
on top of that.

The character devices are quite slow and setting multiple lines in sync is
a PITA.

cwassert, 2020-12-08:
Even with `CONFIG_STRICT_DEVMEM=n` it does not run on aarch64 (rpi4, arch linux).
I don't see the EPERM error in `strace -ff` or anything else suspicious. It just hangs and only `kill -9` can make it exit.

cwassert, 2020-12-08:
Have you considered writing a kernel module or using eBPF to update the GPIO pins?

quantenProjects, 2021-02-28:
@cwassert you may want to try this: https://github.com/hzeller/rpi-rgb-led-matrix/issues/812#issuecomment-787527063

closed 2025-11-27 by marcmerlin