hzeller/rpi-rgb-led-matrix #782: Hub75 P10 1/2 Scan 16x32 Outdoor Panel Setup?

nospamsam2011, 2019-03-24:
Good day All,
Can anyone provide some direction as to how I can modiffy Henner Zeller's excellent library/api to work with a yet another odd ball LED Panel?  My panel is a P10, 1/2 Scan, 16x32 Outdoor panel with the HUB75 Interface.  My HUB75 interface includes the usual signals except that A and B are only indicated.  I tried every combination of Demo's command switches and was unable to get any of the demo images to display properly.   Reviewing the doc I see that there is no mention of 1/2 scan and so I suspect that my panel is yet another version... So...I reversed engineered the display to decode how the various control signals affect the individual LEDs.  the results are as follows:
1. Address lines A and B
- Only Address A is used.  Address B must be Low
- Where A=0, Every odd row is displayed and when A=1 Every even row is displayed

2. The LED positioning is really odd... at least to me.  The LED's appear to wrap to the next line every 8 LEDs and this continues across the Columns.  

Attached are a few pics of the panel and the HUB75 connector
[image]

[image]

As an example, I have included a PDF of the LED positioning.  

[P10 16x32 1_2 Scan Outdoor Decoding.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/3000640/P10.16x32.1_2.Scan.Outdoor.Decoding.pdf)

These positions are referenced as from viewing the panel head on with the "HUB75 data in" located on the right side.   Row 1 Column 1 is the top right Corner.  here I get the following:
- LED 0 (R1-G1-B1) (A=0) is located at Row 3 Column 1
- LED 7 (R1-G1-B1) (A=0) is located at Row 3 Column 8
- LED 8 (R1-G1-B1) (A=0) is located at Row 1 Column 1
- LED 15 (R1-G1-B1) (A=0) is located at Row 1 Column 8
etc.

With this arrangement to display a single row one would need to clock 128 times and only set the corresponding bits:

e..g To display Row 1 (A=0), then a total of 128 clock pulses are needed with:
- Clock's 8-15, 24-31, 40-47, 56-63 with a suitable value of R1-G1-B1, all the other clock's will have the R1-G1-B1 set to 0 (off).
- Next Set A=0 and Latch, and then Set OE low

With that said, I am still a newb to the library and so any suggestions on how to modify to use with this panel would be greatly appreciated.

Thanks in advance!

Cheers,

Sam

tammaaziz, 2019-06-23:
Hello,

I do have a similar issue with a P10 panel !!
Any help please

Robycop58, 2019-09-15:
Hello, same problem.
The multiplexer HUB75 IC is 5020. I had try all the combination of multiplexing, scan-mode and row-addr-type options but the pannel don't work properly.
Please hzeller up and help us.....
Thank's

kasperv79, 2019-10-26:
Any luck with this panel? I’m also stocked...  :(

nospamsam2011, 2019-11-03:
Sadly, no progress or luck...That being said I have not had time to dig deep into this issue, as I was hoping that there would have been some info on how best one could tackle the problem.

felaniMT, 2021-07-31:
Hi, it seem this thread still open..i also facing problem with p10 rgb 1/2 scan panel.. anyone have any idea how to check scan pattern?

WernerFettke, 2021-08-18:
Hi All,

Also having the same problem, also tried hours of command line options... the best I can get is (using the word 'hello' as my test
[image]
):

which is the result of:
sudo ./text-example -f ../fonts/8x13.bdf -C0,0,255  --led-slowdown-gpio=4 --led-parallel=1 --led-rows=16 --led-cols=64 --led-chain=1 --led-cols=32 --led-multiplexing=9

And at a different font size: sudo ./text-example -f ../fonts/6x10.bdf -C0,0,255  --led-slowdown-gpio=4 --led-parallel=1 --led-rows=16 --led-cols=64 --led-chain=1 --led-cols=32 --led-multiplexing=9
[image]

The panel model number is: p10-2s-32x16-3535-v1 (1/2 scan)

I saw some articles suggesting creating a transform file, but I have not had any luck with that either (might be related to I'm not 100% code efficient).

If anyone can assist I would really appreciate it, even if you could just point me in the right direction.

Thanks in advance.

davemaster, 2021-08-19:
When I try this great code for the first time, I had the adafruit panel (1/16) and other from china, a P10 but with 1/4 scan. check the option "  --led-multiplexing   ", the default one worked perfectly with the Adafruit panel, but for the chinese P10 32x16 1/4 scan, the option what I found by try-and-error was --led-multiplexing=8.  Now, your call for test it with yours

Best regards

WernerFettke, 2021-08-19:
Hi Dave, Many thanks for the feedback, I've tried using the multiplexing option already, I found the best result was option 9 (9=P10-128x4-Z), however the images in the original post are using this already. It's almost as if I'm almost there.... just need a small tweek somewhere. In terms of the chinese panels you using, are they also p10-2s-32x16-3535-v1 ?  Did you need to make any other changes to your .h or .cc files in order for the panels to work ?

davemaster, 2021-08-20:
Because I am using now a RPi 4, the slowdown option must be set with 2
minimum, try that too, check the rows, cols (those were my errors at the
beginning)

marcmerlin, 2025-09-29:
thanks for helping resolve this old bug @pataelmo

closed 2025-09-29 by marcmerlin