hzeller/rpi-rgb-led-matrix #783: Flickering only with my code

ggilestro, 2019-03-24:
I am using the Adafruit bonnet to drive some 64x64 LEDs.
Everything works perfectly fine with the provided samples but for some reason, I get quite a decent amount of flickering with my code (found here: https://github.com/gilestrolab/rPI-buridan/blob/master/buridan.py )

My code creates PIL images by drawing shapes on them and then sends its image to the display to create multi-layered animations. I have no idea why I get flickering. I have tried the workaround mentioned here but to no avail.

Can someone please help? Thanks a lot!

angelogoncalve, 2019-03-27:
Try the option to reduce the flicker:

options.pwm_bits = 1

or kill the thread that is running, when you call a new one.

https://www.oreilly.com/library/view/python-cookbook/0596001673/ch06s03.html

[Animator.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/3013645/Animator.zip)

angelogoncalve, 2019-03-27:
The kill thread method must be called in each method that are called in the run() method, to not have message above in another one causing the flicker also in the panel/display.

jaygarcia, 2019-05-07:
@ggilestro have you configured your raspberry pi to only use a single core via adding `isolcpus=3` to `/boot/cmdline.txt` ?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828