hzeller/rpi-rgb-led-matrix #789: would it be possible to refresh separately two part of the screen ?

BENETNATH, 2019-03-28:
Hello,

i've got two 64x64x panels and i would like to use the bottom part of them to display clock and date (128x32 size), and upper part to display gif 128x32 with various length

Does it seem possible ?
How should i proceed ?

Thanks

hzeller, 2019-03-28:
you need to treat it as one big framebuffer, but if course nothing prevents you to put different things on different parts of your franebuffer.

BENETNATH, 2019-03-29:
ok, good to know that is feasible.
my limited knowledge is restricted on perl and python..
I'll try to do the monkey and trial/errors with demo codes may lead me to a solution ^^
thanks again for this great tool, i wish i had more spare time to raise my skill to use it properly

closed 2019-03-29 by BENETNATH