hzeller/rpi-rgb-led-matrix #790: RPI 3B+ 8 X modules P5 + PowerSupply 5V60A + Adapter passive 3 = Fickering

gamukeat, 2019-04-02:
Hello,
I do not understand why?
These is my command line options, I use Raspbian list (without GUI) isolcpus = 3 is enabled.
I tried all the combinations but I still have the same problem.
I tried to change Makefile "-DFIXED_FRAME_MICROSECONDS=XXXX", improve little bit, reduce the flickering but the problem is always the same.
I Can't use gpio-slowdown to 1 or 0.

--led-rows=32 --led-cols=32  --led-chain=8 --led-parallel=2 --led-multiplexing=2 --led-pixel-mapper "U-mapper" --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds [100-300] --led-scan-mode=[0,1] --led-pwm-bits=[8,9,10,11] --led-pwm-dither-bits=[0,1,2]

https://www.ali-demir.com/video-1554219170.mp4

you will see the refresh of the video on the door, is not stable.

Thank you in advance,

jaygarcia, 2019-05-07:
hi @gamukeat, have you tried a more optimized distro like DietPi?

leoceto, 2020-11-22:
I do have the exact same behavior on the RPI 3B+, I struggle to find the right parameters while it is working well on RPI2 et RPI4.

Any ideas to optimize it on RPI 3B+ ? It looks like it is between gpio-slowdown 2 and 3 or 1 or 2....

jaygarcia, 2020-11-25:
@leoceto did you follow the above `isolcpus = 3` param?  The flicker depends on a ton of factors, including things like locks to any threads you might be running to feed the buffer for this library as well as overheating. Is your 3B+ actively cooled?

leoceto, 2020-11-26:
Hello, yes i put the isolcpus = 3 param, I manage more or less to have almost no flicker by decreasing the pwm-bits and increasing the pwm-dither-bits while limiting the refresh rate with a gpio-slowdown = 2, but that's a pity to decrease the performance for that....

closed 2025-11-27 by marcmerlin