hzeller/rpi-rgb-led-matrix #791: Python - Changing —led-brightness in real time

glapadre, 2019-04-03:
Hey so I’m trying to switch the LED brightness depending on what time of day it is.. I leave the LED display running constantly as a dashboard and want to turn it down at night ( say 10:00pm ). I’ve tried so many different things and nothing seems to work..

I always end up with a “FYI: Can’t set realtime thread priority=99 Operation not permitted” error. 

I’ve tried changing hzeller’s samplebase.py code and and also re-running process(self) but can’t get it to budge.

Any ideas out there or has anyone done this? Thanks a lot.

angelogoncalve, 2019-04-06:
to turn tjhe panel down you can use the clear() method

glapadre, 2019-04-06:
> to turn tjhe panel down you can use the clear() method

Can you explain this in more detail if possible?

angelogoncalve, 2019-04-06:
from rgbmatrix import RGBMatrix, RGBMatrixOptions

# Configuration for the matrix
options = RGBMatrixOptions()
options.rows = 32
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'regular'  # If you have an Adafruit HAT: 'adafruit-hat'

matrix = RGBMatrix(options = options)
matrix.Clear()

angelogoncalve, 2019-04-06:
#!/usr/bin/env python
from samplebase import SampleBase

class GrayscaleBlock(SampleBase):
    def __init__(self, *args, **kwargs):
        super(GrayscaleBlock, self).__init__(*args, **kwargs)

    def run(self):
        max_brightness = self.matrix.brightness
        count = 0
        c = 255

        while (True):
            if self.matrix.brightness < 1:
                self.matrix.brightness = max_brightness
                count += 1
            else:
                self.matrix.brightness -= 1

            if count % 4 == 0:
                self.matrix.Fill(c, 0, 0)
            elif count % 4 == 1:
                self.matrix.Fill(0, c, 0)
            elif count % 4 == 2:
                self.matrix.Fill(0, 0, c)
            elif count % 4 == 3:
                self.matrix.Fill(c, c, c)

            self.usleep(20 * 1000)

# Main function
if __name__ == "__main__":
    grayscale_block = GrayscaleBlock()
    if (not grayscale_block.process()):
grayscale_block.print_help()

angelogoncalve, 2019-04-06:
[pulsing-brightness.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/3050806/pulsing-brightness.zip)

glapadre, 2019-04-07:
Thank you @angelogoncalve !! This is exactly what I was looking for :)