hzeller/rpi-rgb-led-matrix #794: Auto restart python script

brilt, 2019-04-04:
Hello, at scool i'm using 2 led matrix panel, both are 32x64. I have to display text wich is written in file. I've done it but i have to restart the program (called test.py) myself (by doing ctrl + c and then enter sudo ./test.py). I want to automate this program so that i don't have to use ctrl + c and sudo ./test.py. I'm a begginer at using python so I don't know what I can use. Here is the program : 
[image]

hzeller, 2019-04-04:
So you want to change the text whenever the textfile changes ?
So instead of stopping and starting, you should use the stat() call to see if the file changed, then reload it. Or simply reload it every time within your loop.

Also make sure to not waste CPU: right now, you are outputting the same text in a tight while loop, which is not really needed. You only really need to do the clear/drawtext/swap whenever the content of your text changes. So as a simple solution I suggest to have an infinite loop, which has a sleep of 0.5 seconds or something, then read the file into the string and display it. Thus you only do something every 0.5 seconds.

Ask a local friend who knows Python to help you with that.

brilt, 2019-04-06:
So, i did the loop that read the file every 0.5 seconds and display it and it works, thank you a lot !

closed 2019-04-06 by brilt