hzeller/rpi-rgb-led-matrix #795: OverflowError: can't convert negative value to size_t

northwarks, 2019-04-13:
Hi Guys -I'm having issues with SetImage(), I can display text with no issues at all, as soon as I try to add an image I get the following error:

```
  File "matrix_v2.py", line 134, in DisplayUpdate
  mycanvas.SetImage(image, 0, 0)
  File "core.pyx", line 22, in rgbmatrix.core.Canvas.SetImage
  File "core.pyx", line 46, in rgbmatrix.core.Canvas.SetPixelsPillow
  OverflowError: can't convert negative value to size_t
```

I'm calling DisplayUpdate() from a thread every second, no other thread or function is writing to the matrix. I'm a little unsure if I'm _actually_ understanding the correct way of implementing images but I've tried for hours and it always boils down to the `OverflowError: can't convert negative value to size_t` error despite PIL reporting the image as 50x50 RGB. 

```
  def DisplayUpdate(self):
    mycanvas = self.matrix.CreateFrameCanvas()
    #local_weather={'mylocal_temp': None, 'mylocal_humid': None, 'mylocal_status': None, 'mylocal_icon': None}
    #local_time={'curr_day': None, 'curr_time': None, 'curr_sec': None}
    _sec = local_time.get('curr_sec')
    _day = local_time.get('curr_day')
    _time = local_time.get('curr_time')
    _temp = local_weather.get('mylocal_temp')
    _humid = local_weather.get('mylocal_humid')
    _status = local_weather.get('mylocal_status')
    _icon = local_weather.get('mylocal_icon')
    self.matrix.Clear()
    graphics.DrawText(mycanvas, self.font1, 10, 20, self.blue, _sec)
    graphics.DrawText(mycanvas, self.font3, 34, 40, self.blue, _day)
    graphics.DrawText(mycanvas, self.font1, 38, 30, self.blue, _time)
    graphics.DrawText(mycanvas, self.font3, 3, 60, self.blue, _temp)
    graphics.DrawText(mycanvas, self.font3, 55, 60, self.blue, _humid)
    graphics.DrawText(mycanvas, self.font3, 100, 60, self.blue, _status)
    image = Image.open(_icon)
    image = image.convert('RGB')
    image.load()
    mycanvas.SetImage(image, 1, 1)
```

Any pointers or skeleton code would be really appreciated :)

hzeller, 2019-07-07:
it looks like you are createing a FrameCanvas every time you call DisplayUpdate ? This will allocate a new frame every time so you will run out of memory.

You should only create a frame canvas once, then swap on vsync.

As with the `size_t` error: can you figure out where that message is coming from ? It looks like somewhere from the inside of the generated Python-wrapper.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin