hzeller/rpi-rgb-led-matrix #796: 64x64 Matrix (32s) without a D line?

jaygarcia, 2019-04-16:
Hello all,

First I want to say a huge **THANK YOU** to those of you who contributed to this project. It's made my project quite easy to get rolling. I had some 64x32-P6 Matrices lying around and prototyped a video display using a 3B+ and the active board from Electrodragon. It works so frigging well! (Proof of concept here: https://www.youtube.com/watch?v=eyJVVf5jL0k)

To move the project along, I ordered some 64x64 Panels from China `P3(2121)64x64-32S-M1` and ran into a situation that is unfamiliar ground for me.  The panels have `E` at Pin 8 but don't have a `D` line, but rather a `GND`.  Has anyone seen this before?

Here's an image of the J-OUT header.  Should I just clip the wire that leads to pin 12, so I don't create a short to ground from the 5V buffered `D` line?

[image]

Any help would be greatly appreciated =).

jaygarcia, 2019-04-16:
Using this issue to document what I've learned, so hopefully anyone else who finds themselves w/ these panels can follow and resolve their issues.

Spark Fun is the only place I've been able to find such a panel via the [SparkFun RGB Panel hookup guide article](https://learn.sparkfun.com/tutorials/rgb-panel-hookup-guide/all)
[image]

I read the following article and I highly suggest anyone who is new to this (like I am) do the same: [Everything You Didn't Want to Know About RGB Matrix Panels](https://www.sparkfun.com/sparkx/blog/2650)

-----
*Unless I am mistaking:* This is a 1/32 panel, meaning it will scan 1/32 of the pixels at a time (per address combination). 1/32 of a 64x64 display (total 4096 pixels) is 128 pixels long (two rows of 64 pixels wide). 

The maths looks like this:
```
// JavaScript 
var panelWidth = 64,
    panelHeight = 64,
    totalPixels = panelWidth * panelHeight, //4096
    scanRate = 1 / 32,
    numPixelsPerScan = totalPixels * scanRate;

console.log(`numPixelsPerScan = ${numPixelsPerScan}`); // numPixelsPerScan = 128
```

This panel only has four addressable lines. That's four bits of values, giving us a total of 16 possible rows to address.  This is where I get caught up. At 1/32 scan, the board is divided by two, but we can only "Address 16" rows. (Rows 1 - 15 and 32 - 47).  How can I address the 2nd half of the row sets? (17 - 31, 47- 64) without a D line? Seems hat most 64x64 panels have 5 addressable lines (A,B,C,D,E) #754 .

I've contacted the manufacturer (Hoonshine) for a data sheet. This is the [panel I purchased](https://www.alibaba.com/product-detail/SMD-full-color-3in1-rgb-led_60839777344.html)

jaygarcia, 2019-04-16:
Pulled out a Multimeter to see if Pin 12 (GND -- Missing D) pin is actually a ground and it turns out that there is no continuity between Pins 12 and 4 or 12 and 16, however Pins 4 and 16 are actually ground. I'm beginning to wonder if Pin 12 is Not connected or actually a D line and improperly labeld?

jaygarcia, 2019-04-16:
Pulled out a Multimeter to see if Pin 12 (GND -- Missing D) pin is actually a ground and it turns out that there is no continuity between Pins 12 and 4 or 12 and 16, however Pins 4 and 16 are actually ground. I'm beginning to wonder if Pin 12 is Not connected or actually a D line and improperly labeled?

jaygarcia, 2019-04-16:
### Success!!
Pin 12 must be a `D` address line.

Single chain (P0)
```
./demo --led-rows=64 --led-cols=64 -D 9
Size: 64x64. Hardware gpio mapping: **regular**
```
[image]

```
./demo --led-rows=64 --led-cols=64 --led-chain=2 -D 9
Size: 128x64. Hardware gpio mapping: regular
```
[image]

closed 2019-04-16 by jaygarcia

jaygarcia, 2019-04-16:
Wanted to show how well this library works. 
The pixel data is pumped via TCP/IP and then written to the frame buffer for display. 
The flicker seen is not visible in person. 

https://youtu.be/ODnbhvbLX9E

angelogoncalve, 2019-04-17:
Dear Jay Garcia, nice and a good work. Can you disponibilize the code to test in a panel 128x32 chain 2 units (2 x 64)x32, that I am using the 2 panels conected to a RPi3 (the server always on/up) and a rgb matrix board (blue board) in a bus car please, for sign destinations?

Thank you for your atention.
Ângelo.

jaygarcia, 2019-04-17:
Hi @angelogoncalve are you asking specifically for what I'm doing with my project ( Display data via `TCP/IP` using this library ) or are you asking for generic code to use 2 x (64x32) Panels like in [my first video](https://www.youtube.com/watch?v=eyJVVf5jL0k)?

angelogoncalve, 2019-04-18:
Hi Jay. It is the one for display data via TCP/IP using this library, if you can share the code for that project please?

Thank you for your atention.
Ângelo.

jaygarcia, 2019-04-18:
Hi @angelogoncalve , 

It will take me some time to make it truly workable library with proper threading & exception handling, but I'll provide some gists in the meanwhile. The `Receiver` runs on the pi in my video and a modified version of the `Sender` runs on the PC. 

`Receiver (runs on the RPi):`
https://gist.github.com/jaygarcia/c8b0ce341b36c453c869fc2fe8c4f29c

`Sender (runs wherever)`
https://gist.github.com/jaygarcia/c8b0ce341b36c453c869fc2fe8c4f29c

You will absolutely need to install lib boost (1.7.0). http://www.linuxfromscratch.org/blfs/view/cvs/general/boost.html

angelogoncalve, 2019-04-19:
Hello Dear Jay. I wait for your TCP/IP version workable library, with the manager for threads requests and exception calls.

Thank you.

The sender gist have the same link as the receiver?

Thank you for your atention.

Best Regards,

Ângelo.

jaygarcia, 2019-04-19:
@angelogoncalve my apologies. I've updated the link in the above post (it was in the receiver comment all along).

https://gist.github.com/jaygarcia/5d77e6687e742dcfb19f58728c997b76

angelogoncalve, 2019-04-19:
The sender and the receiver can be possible be mada using Python language as well?

Thank you Jay for your help.

jaygarcia, 2019-04-19:
I imagine it's possible, though I am not a Python programmer at this time.

angelogoncalve, 2019-05-02:
Hi Dear Jay. I hope you are fine.

Did you can share the full program/code for led panel/display 64x64 with scan rate 1:32, like the exemple in the vídeo link (https://youtu.be/ODnbhvbLX9E) with the funcionality for TCP/IP version workable library, with the manager for threads requests and exception calls?

Thank you for your atention and help.

Best Regards,
Ângelo.

jaygarcia, 2019-05-02:
Hi @angelogoncalve i'm doing well thanks :)! Hope you are as well.

I'm currently working on a suite of classes that do this, but I'm learning proper concurrency at the same time.   The project is quite ambitious but functional -- just needing optimization.  

I'm going to explain what I'm doing because it may not be exactly what you are exactly looking for.  I would hate for you to wait endlessly as I get this stuff done.

### Project idea
- 1 x Rock64 (or RPi3b+) (Controller & `Master`). Runs my video game, audio, blue tooth controller. The video buffer that would normally be piped through SDL gets piped to a network display class that splits the buffer into 5 "strips" of video and sends them over TCP/IP to the  5 x RPi3B+ `slave` devices.  Each "strip" has it's own dedicated thread to pipe the TCIP/IP data. (this is where I think my optimizations need to happen).
- 5x RPi3B+ `slave` devices take the buffer data received via TCP/IP and renders them via the `rpi-rgp-led-matrix` library. 

```
# Each panel is 64x64 pixels (p3 displays sourced form Alibaba.com via HoonShine)
# Each column is a "Strip" of display data controlled by a single "Slave" RPi3B+
# Boxes below represent displays. Each "box" coordinates are [ `column` . 'row` ]. 
[ 1.1 ][ 2.1 ][ 3.1 ][ 4.1 ][ 5.1 ] 
[ 1.2 ][ 2.2 ][ 3.2 ][ 4.2 ][ 5.2 ] 
[ 1.3 ][ 2.3 ][ 3.3 ][ 4.3 ][ 5.3 ] 
[ 1.4 ][ 2.4 ][ 3.4 ][ 4.4 ][ 5.4 ]
```

I have the rpi's currently receiving TCIP/IP data via an 8port gigabit switch and my current situation is audio is "popping" on the rock64, which is an indication that something's not optimized properly w/ the threading and/or buffering of audio data.

Again, I don't know what you're looking to do, but I will promise to open source everything once I'm done.  I'm currently also fighting build configuration issues w/ our game's codebase for the ESP32 target (our game was written for the ODROID GO and will soon be open sourced as well).  Here's a WIP demo of the game running on SDL: https://www.youtube.com/watch?v=De_emh_Cslw

angelogoncalve, 2019-05-05:
Hello Dear Jay.

Thank you for your atention.

The project is similiar is via web page or by phone/server/master sending messages for led panels/clients/slaves using a RPis for each one via TCP/IP or JSON RPC command.

Thank you for your help.

angelogoncalve, 2019-05-11:
Hi Jay. 

Could yuou share the update for your Raspberry Pi-driven RGB matrix panel display experiment?

Thank you.

jaygarcia, 2019-05-11:
Hi @angelocalve, it’s going well. Getting close to build. I need to modify my implementation of @hzeller’s library so that it automatically spawns a process and will be blogging about this as well as producing some videos on YouTube. 

To use this code, you’ll need to know about writing to bitmap buffers and concepts like threads. 

Here is an image of the code working on all 5 raspberry pi’s at the same time. 

https://www.instagram.com/p/BxVNuZIgacq/?utm_source=ig_share_sheet&igshid=9k6aukjfrjsg

jaygarcia, 2019-05-11:
@angelocalve
https://youtu.be/mpJB9lXEfg8

hyamanieu, 2019-05-11:
Hello @jaygarcia , very nice work! We're doing a similar project (https://github.com/murapixrepo/murapix) and we're thinking of going large scale as you did. How come you're needing 5 RPis 3B+ to run the panels? Our frame rates are about similar from 1 to 6 panels and hzeller claims we can link up to 3*12 panels without too much trouble.

Thanks!

angelogoncalve, 2019-05-12:
Hi Jay. Thank you. 
The video game is conected to RPi3 that send/write IN bitmap/data buffer to 5 RPis by TCP/IP.
You can control the 5 RPis/playing the video game using a command by TCP/IP.
The thread you use to control when the image to put in the 5 RPis is not fixed, but yes is like a video or a scroll text.
Yes we can link up to 3*12 panels in only one RPi, but it is necessary more current to supply all the 36 displays.
I wait for your sharing the code Jay when you finished the implementation modifies in hzeller library.
Thanks for your help.

jaygarcia, 2019-05-12:
Hello @hyamanieu and @angelogoncalve thank you for the heads up on the number of panels the Pi and this library can handle. I somehow missed that! 

Would we be able to get 60fps with that many panels? I need 20 for this project and the reason I split it up was for speed purposes. 

- TCP/IP data chunks are smaller
- spread the cpu workload across different pi’s. (This is where I am likely wrong)

As far as power (electricity), no issues there. I have the power supplies needed :). 

If a single pi can be used as a slave, the complexity of this project dramatically gets reduced :)

angelogoncalve, 2019-05-12:
Hi Jay.

Hello Jay.
I think instead dividing the data/buffer using RPis and using threads in these ones, for 20 panels you will get 60fps using only a single pi as a slave, but the TCP/IP data are bigger for one RPi, with only one buffer with all the bitmaps to play in the all displays. But can be done as you are doing using 5 slaves/RPis, using more code and more equipment/hardware.

jaygarcia, 2019-05-12:
Thanks @angelogoncalve. I will see if I can pivot to using a single Pi for latency testing. :). 

I will have to refactor my code to lay out the buffer data appropriately.

krishnasandeep10694, 2019-05-14:
Hi @jaygarcia ,
I am planning to use this library with rock64. I have read in couple of sources that rock 64 and rpi has similar gpio pinout. But whem i am trying to run a runtext program which is in samples directory I am getting below error even though i am running as sudo.

```
rock64@rock64:~/Downloads/rpi-rgb-led-matrix-master/bindings/python/samples$ sudo python3 runtext.py --led-chain=1
mmap error: : Operation not permitted
RPi1: MMapping from base 0x20000000, offset 0x200000
Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command
Press CTRL-C to stop sample
Segmentation fault
```
I am running Ubuntu_18.04_Bionic_LXDE_Desktop on rock 64. I dont have much idea about hardware memory mappings. I did a bit of google research on the above issue and people are mentioning to tweak some flags in kernel config and that would be dangerous. 

You mentioned in one of the comments that you are using an rock64. So i am posting it here.

Any help would be appreciated.

Thanks.

jaygarcia, 2019-05-16:
@angelogoncalve, things are starting to shape up. Still more work to do: https://github.com/jaygarcia/network-rgb-matrix-display/blob/master/md/How_it_works.md

jaygarcia, 2019-05-30:
Quick update for @angelogoncalve and @hyamanieu.

I ran an experiment where I connected 10 panels to a single Pi using 2x multiplexing.  I was less than 60 frames / second. The display on demos flickers badly. :(

Running 4 panels per pi.  Single chain (1 x 4 panels), ~67Hz. Dual chain (2 x 2 panels) ~88Hz.

jaygarcia, 2019-06-12:
For anyone following, we assembled the prototype and it can be viewed here:
https://www.youtube.com/watch?v=s0lbVMcgfww

Next step is to document the project & open source. :)

Beersatron, 2019-06-12:
Looks awesome! I would be interested in how you built the frame and wired the power, if you have the time to document and share that too.

jaygarcia, 2019-06-12:
absolutely @Beersatron .  Long story short, i'm using lib boost :P . Everything will be open sourced.

jaygarcia, 2019-06-12:
duh. sorry, was thinking "Frame" as in video frame. Yes!! aluminum frame.  That was using products from 8020.net and "flat bar" from home depot. But yes, build instructions are in the backlog.

Beersatron, 2019-06-12:
@jaygarcia Great! I used perforated flat bar from home depot and some 2x4/2x6 lumber for my scoreboard and it looks ok but not as sharp as what you have managed to produce. I did try extruded aluminium for a smaller, earlier, project but that also never looked quite right.

jaygarcia, 2019-06-12:
Our issue is figuring out how and where to mount the various hardware & power systems:
- 2x power supplies
- 1x USB Power block
- Network Switch
- Speakers, etc..

We've still got a way to go, but it's at least functioning. :)

FFFFFF12ABB17, 2019-12-22:
I have this same board and have issue with display.  I am using the rpi bonnet.  How did you solve this, exactly?  Simply with the command line displayed above?  It did not work for me.

jaygarcia, 2019-12-22:
@FFFFFF12ABB17 are you sure D is the address line?  I am not using the Rpi bonnet, and thus am unfamiliar with that board. 

“Does not work” isn’t a description I can do much with. 

For those following this thread, the final build has been working for some time and will be doing the full write up after the holidays. 

[image]

[image]
[image]

MusienkoAnton, 2021-05-03:
@jaygarcia Hello
I try to use 64x64 modules in my project and see the same issue as you wrote at start.
I only see 16 pixels, than 16 nothing, than 16 see and than 16 nothing.
Also I see GND on label of port on 12 position.
I`m new with boards of 64x64.
Please help me explain more what you do make it working. I`m using Raspberry and python.