hzeller/rpi-rgb-led-matrix #797: Give support forum in README.md and possible source for adapter/active-3

marcmerlin, 2019-04-17:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md
is pretty detailed but does not mention anything on where to get help/ask questions.
Then people due stupid things like opening an issue to ask for help when it's not actually an issue, just like I did here :)

I'd love to get a premade adapter/active-3 but honestly it's more work than I have time and skill for. I'm sure some people who are better setup would be happy to make some and sell them. Looks like I already found one here https://github.com/hzeller/rpi-rgb-led-matrix/issues/763 although shipping from Europe might cost as much as the board :)
Would be great to have a list of folks who are willing to sell them.
In the US, tindie is great for that, for example https://www.tindie.com/products/jasoncoon/16-output-nodemcu-esp32-wifi-ble-led-controller/

jaygarcia, 2019-04-18:
@marcmerlin  - Great points! You can buy a pre-made adapter from [electro dragon](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/). I have ordered 7 of them and they work very well!  (See this video i posted a few days ago) https://www.youtube.com/watch?v=ODnbhvbLX9E

Please consider a PR that adds the above information to the repository :).

hzeller, 2019-04-18:
Let me see if I can set up a tindie

marcmerlin, 2019-04-18:
@jaygarcia thanks for the link, just ordered a few. Shipping is several times the price of the board of course, but no biggie :)
@hzeller tindie would be great indeed.
And yes, happy to send a PR if that helps, once we know what the best information to put in, is.

jaygarcia, 2019-04-18:
Awesome @marcmerlin . keep in mind, if you have a panel that has an E address line, you'll need to jumper pin 4 or 8.  There are issues open and closed that demonstrate how to do this.

marcmerlin, 2019-04-18:
@jaygarcia  yes, I saw that in the documentation, thanks.

marcmerlin, 2019-04-18:
As discussed with @hzeller offline, encouraging people to use github issues for help as opposed to issues with the code, is likely not the best plan, and at least something many other github project explicitly discourage.
From having worked with FastLED, ESP32, and SmartMatrix, here are possible support forums that you could use. 

1) I like discourse as used by SmartMatrix: https://community.pixelmatix.com/ . It costs a few dollars per month to get your own instance I think, or maybe Louis from SmartMatrix would be willing to add a few topics for this project on his instance if you don't mind the domain name (SmartMatrix is the best RGBPanel driver available for teensy/ESP32, but it sadly lacks support for newer panel chips and AB addressing)
2)  https://discordapp.com would be another option
3) FastLED moved to reddit, even if i'm not personally a fan: https://www.reddit.com/r/FastLED/

Separately from one of the above 3, ESP32 uses gitter for more real time help between users (depends on volunteer and a community to support it): https://gitter.im/espressif/arduino-esp32
I'd say gitter is optional but an option.

Let me know if you'd like me to ask Louis whether he'd be ok hosting this project on his discourse instance, should you want to try that route (or you can request/get your own)

jaygarcia, 2019-04-18:
@marcmerlin this makes a ton of sense! real-time comms (discord) is always dependent on who is available, listening, etc.. It becomes an issue where you want to post a question, but can't stay on to see who responds, etc..  Forums / reddit do make sense. 

+1 for discourse! Many other communities use it, like the Arduboy community. https://community.arduboy.com 

Funding for the hosting and other costs can be done via donation.  I know the CEO (Pia Mancini) of Open Collective: https://opencollective.com

Small aside: The SmartMatrix project is awesome. I used the branch that supports ESP32 to prototype my project, but landed on this library as the Pi3B+ is way faster & I didn't have to do any funky things (custom ESP32 boards, etc..) to make publishing buffer data via TCPIP possible.

marcmerlin, 2019-04-18:
@jaygarcia  yeah, happy to donate a few dollars for discourse, it's a nice interface and worth it IMO.

Off topic:
yeah, ESP32/SmartMatrix works but I had to use an unrelated shield (Jason Coon's neopixel level shifter tindie shield mentioned above and patch a wire to keep serial debugging working).
It however tops out around 128x64 due to DMA memory shortage issues and how SmartMatrix is implemented (it favours display quality over memory usage). 
I wrote some tips on how to use SmartMatrix with ESP32 here: http://marc.merlins.org/perso/arduino/post_2019-04-01_SmartMatrix_-SmartMatrix-Shield-v4-for-Teensy_-ESP32-shield-with-level-shifter_-and-SmartMatrix_GFX.html#ESP32
That's the project I just finished but that is stalled at 96x64 (still not bad) and why I'll have to switch to rpi-rgb-led-matrix to go higher rez: http://marc.merlins.org/perso/arduino/post_2019-04-08_Clubbing_-EDM-Festival-and-Burning-Man-LED-Pants-and-Shirt-v4-on-ESP32-with-RGBPanels-and-SmartMatrix_GFX.html

marcmerlin, 2019-04-19:
@jaygarcia added the seller you recommended in this PR: https://github.com/hzeller/rpi-rgb-led-matrix/pull/799

embedded-creations, 2019-05-03:
Not sure if this project qualifies, but if so, can't beat free: https://blog.discourse.org/2018/11/free-hosting-for-open-source-v2/

FYI I pay for and am happy with discoursehosting.com for SmartMatrix Community.  I chose it a few years ago, and it was the only affordable managed option when I looked back then.  $5/month for a Digital Ocean instance and your time to set it up and maintain Discourse is another cheaper (if time is free) option.

marcmerlin, 2019-05-03:
Thanks @embedded-creations  I didn't even know about that plan, and given that this project has 41 contributors, it should qualify

marcmerlin, 2020-01-30:
@jaygarcia @hzeller @embedded-creations : so, the active-3 board seller is now mentioned in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter
At that pricepoint, no one is (understandably) interested in competing.
However, I would very much like an active or (or will settle for passive) board where the HUB75 plugs are pointing sideways, as close to the Pi underneath as possible.
The idea is to have the entire thing be as flat as possible.

Can someone with more hardware routing/ordering foo than me design an updated board where the HUB75 plugs are parallel to the Pi (as close as possible) instead of pointing up?

@embedded-creations  you may also be interested in https://github.com/hzeller/rpi-rgb-led-matrix/issues/972 which so far seems to hint that running on a Pi0 has performance issues. They might be solvable, but they are a concern for now (hence my switching back to rPi3).

@hzeller with active-3 board, I noticed that the Pi serial port isn't usable anymore since it's used for IO. Are there any 2 IO pins left at all on the Pi to do some kind of serial, or nothing left?

hzeller, 2020-01-30:
Unless you use more than 2 chains, the TxD is kept free.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L61

And the RxD is also kept free unless using 1:32 multiplexing panels.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L41

The library will only allocate the pins it actually needs, so the rest are free for whatever else they might be useful for.

There are no officially supported pins left. Well, technically, Pin 27 and Pin 28 could be used as GPIO, but the Pi specification does not want that (but I think I confirmed that they actually could be used). So you could hack something where you change the pinout accordingly.

marcmerlin, 2020-01-30:
Thanks @hzeller, sounds like that's a good reason for me to use those ABC 128x64 panels over the ABCDE ones, even if they are not really faster in my testing, they only use 3 address lines. Unfortunately, this doesn't give me back TX when using 3 strings, but at least it gives me 2 pins back (15 + 25).  I just read that software serial isn't really a thing on rPi though. Mmmh....
I guess I could cut a trace on the active 3 board and move .p2_r1 to gpio25 (d) which won't be used in AB/ABC panels.

jaygarcia, 2020-01-31:
If this is some use to folks: We opted to setup daughterboards with a GPIO extender board to allow for better airflow for the dual fans we have.  It's not very stable though :( 

[image]

[image]

marcmerlin, 2020-01-31:
@jaygarcia  sorry, I don't see from your picture. Is that the regular active-3 board?
Which way are the connectors pointing? 
I was trying to get a thinner solution, not a thicker one :)

marcmerlin, 2020-02-12:
@hzeller  this is what I had in mind. i just Emailed electrodragon, and they were very responsive. With a little luck, they'll be able to make an updated version of the board that is lower form factor.  
Would you say that this new format would work better for all/most?
[image]
it would use angled connectors like these (wrong plug, but you get the idea)
[image]

marcmerlin, 2020-02-13:
I made a new bug for the angled board option: https://github.com/hzeller/rpi-rgb-led-matrix/issues/797

marcmerlin, 2020-03-18:
here is what it looks like with the angled connectors installed. Unfortunately board routing does not allow putting the connectors under the board, but they work above if you cut the notch and glue it to the side it used to be on
[image]

I have more details/pictures on this page: http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html

I think we only have one source for active-3 but electrodragon has been great in both price and support. I'm fine with closing this issue.
The only question for electrodragon is
1) should the default board come with angled connectors (probably not because of the notch cutting issue or using plugs without the notch which could get people to plug the wrong way around)

2) would there be a demand for a re-designed board which allows the connector to be underneath at the 90 degree angle like in the picture I attached? (I'd like that, but I may be the only one to care)

marcmerlin, 2020-03-18:
And @hzeller this bring us back to the point that you probably have a big community at this point and (ab)using issues to chat is not the best way.
Short of a web board like mentioned above (what pixelmatix uses at https://community.pixelmatix.com/ ), or discordapp, a mailing list, maybe?

closed 2025-11-27 by marcmerlin

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828