hzeller/rpi-rgb-led-matrix #798: 32x32 led - 3chains 6 matrix row problem

istoony, 2019-04-18:
Hi,
We are a group of students and we are building a led wall by using 3 chains with 6 matrix each. We use a raspberry pi B3+ and the active multiplexer described in this repository.
The problem we are facing when  we run the examples is that half of the rows do not work.
I hope you can help us.
Thank you

[image]

jaygarcia, 2019-04-18:
@istoony, it would be very helpful if you could post the following:

- your demo code usage (command line with switches)
- Information about your RGB panels (refresh rate, connector type)
- how you are connecting them to the PI (Active, passive adapter)?

istoony, 2019-04-20:
Hi, thank you for your message.
We used the active adapter.
The matrix we have are the following https://www.adafruit.com/product/1484
(1/16 refresh rate).
We tried a lot of different codes:
The baseline was:
-->  sudo ./demo -D 0 --led-chain=6 --led-parallel=3  --led-gpio-mapping=regular
And from this we tried all the combination of --led-slowdown-gpio:
0 -> there are some noise at the left of the matrix
1 or 2 -> same result and same problem
We also tried all the combinations of --led-multiplexing but the results are still bad.
We also tried to use --led-addr-type without reaching any good result.
By using --led-show-refresh we reach around 220Hz.

In each configuration we have the odd rows that does not work. For example in the picture there should be the rotating square.
[image]

sudo ./demo -D 0 --led-chain=6 --led-parallel=3 --led-gpio-mapping=regular --led-show-refresh  --led-rows=32 --led-cols=32 --led-row-addr-type=2 --led-slowdown-gpio=1 --led-scan-mode=1

Maybe we forgot something on the active adapter.
We soldered 4 capacitor, 4 74HCT245, 1 resistance and we connected the 4 and 8 pin to GND.

Thank you for your help.

angelogoncalve, 2019-04-21:
Try

sudo ./demo -D 0 --led-chain=6 --led-parallel=3 --led-gpio-mapping=regular --led-show-refresh --led-rows=32 --led-cols=32 --led-row-addr-type=2 --led-slowdown-gpio=2 --led-scan-mode=1 --led-multiplexing=0

istoony, 2019-04-23:
Hi,
We were inexperience. There was a cable in the connection between the active adapter and the matrix that was defective. It was enough to change it.
The strange thing was that the entire wall (the other two chain) was influenced by the problem of a single cable on a single chain.

Thank you for your help.
I close this issue

closed 2019-04-23 by istoony