hzeller/rpi-rgb-led-matrix #800: issues with make and compile

Toootless, 2019-04-20:
I am using your libraries for to drive an RGB-LED 32x 16. I am also using some code from WiringPi to import data from an Arduino to the Raspberry Pi. I tried to use Greany to compile and build the file, but geany has an issue find the LED_matrix library. 

gcc -Wall -o "display3" "display3.c" -lwiringPi -lwiringPiDev -lrt -lm -lpthread (in directory: /home/pi/rpi-rgb-led-matrix-master/examples-api-use)
/tmp/ccmhKKhh.o: In function `main':
display3.c:(.text+0x708): undefined reference to `led_matrix_create_from_options'
display3.c:(.text+0x728): undefined reference to `led_matrix_create_offscreen_canvas'
display3.c:(.text+0x740): undefined reference to `led_canvas_get_size'
display3.c:(.text+0x950): undefined reference to `led_canvas_clear'
display3.c:(.text+0xab0): undefined reference to `led_canvas_set_pixel'
display3.c:(.text+0xc4c): undefined reference to `led_canvas_set_pixel'
display3.c:(.text+0xde8): undefined reference to `led_canvas_set_pixel'
display3.c:(.text+0xf84): undefined reference to `led_canvas_set_pixel'
display3.c:(.text+0xfcc): undefined reference to `led_matrix_swap_on_vsync'
display3.c:(.text+0xfd8): undefined reference to `led_matrix_delete'
collect2: error: ld returned 1 exit status
Compilation failed.

I then tried to use the make file to compile and build the file, but cannot find the correct part to add to the make file to find the library for the WiringPi.

pi@raspberrypi:~/rpi-rgb-led-matrix-master $ sudo make
make -C /home/pi/rpi-rgb-led-matrix-master/lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix-master/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix-master/lib'
make -C examples-api-use
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix-master/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/pi/rpi-rgb-led-matrix-master/lib'
make[2]: Leaving directory '/home/pi/rpi-rgb-led-matrix-master/lib'
g++ display3.o -o display3 -L../lib -lrgbmatrix -lrt -lm -lpthread  -lstdc++ 
display3.o: In function `getData_convertData':
display3.c:(.text+0x2e4): undefined reference to `serialOpen'
display3.c:(.text+0x330): undefined reference to `wiringPiSetup'
display3.c:(.text+0x378): undefined reference to `millis'
display3.c:(.text+0x394): undefined reference to `millis'
display3.c:(.text+0x3d4): undefined reference to `serialPutchar'
display3.c:(.text+0x3f4): undefined reference to `delay'
display3.c:(.text+0x400): undefined reference to `serialGetchar'
display3.c:(.text+0x428): undefined reference to `serialDataAvail'
display3.c:(.text+0x454): undefined reference to `serialGetchar'
collect2: error: ld returned 1 exit status
Makefile:29: recipe for target 'display3' failed
make[1]: *** [display3] Error 1
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix-master/examples-api-use'
Makefile:25: recipe for target '/home/pi/rpi-rgb-led-matrix-master/lib/librgbmatrix.a' failed
make: *** [/home/pi/rpi-rgb-led-matrix-master/lib/librgbmatrix.a] Error 2

Would you know what to add to the make file to add the WiringPi ?

Here is the code.

`/* -*- mode: c; c-basic-offset: 2; indent-tabs-mode: nil; -*-
 *
 * Using the C-API of this library.
 *
 */
// TCP Server
#include "led-matrix-c.h"

#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include <stdio.h>
#include <sys/types.h> 
#include <sys/socket.h>
#include <netinet/in.h>
#include <netdb.h>
#include <string.h>
#include <stdlib.h>
#include <unistd.h>
#include <errno.h>
[... 552 lines cut ...]
					case 15 : // Red
					ln_r = lj_red.mn_r;
					ln_g = lj_red.mn_g;
					ln_b = lj_red.mn_b;
					break;
				}

				led_canvas_set_pixel(offscreen_canvas, i, j, ln_r, ln_g, ln_b);
			}
		}

		offscreen_canvas = led_matrix_swap_on_vsync(matrix, offscreen_canvas);
	}
  } //close( newsockfd );

  /*
   * Make sure to always call led_matrix_delete() in the end to reset the
   * display. Installing signal handlers for defined exit is a good idea.
   */
  led_matrix_delete(matrix);

  return 0;
}
}
`

hzeller, 2019-04-20:
Make sure to link all relevant libraries when you link your program.

Also: don't use wiringPi, it will intefere with the LED matrix as it also attempts to mess with the GPIO pins. It might work, but I'd just use the features already provided in rpi-rgb-led-matrix: If you want to read inputs, see [input-example.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/input-example.cc).

Toootless, 2019-04-20:
Is there another way to read data from an Arduino over the USB on the Rasberry Pi?

Toootless, 2019-04-20:
The part of WiringPi I am using is the serialTest.c I don't thing it does anything with the GPIO.

`/*
 * serialTest.c:
 *	Very simple program to test the serial port. Expects
 *	the port to be looped back to itself
 *
 * Copyright (c) 2012-2013 Gordon Henderson. <projects@drogon.net>
 ***********************************************************************
 * This file is part of wiringPi:
 *	https://projects.drogon.net/raspberry-pi/wiringpi/
 *
 *    wiringPi is free software: you can redistribute it and/or modify
 *    it under the terms of the GNU Lesser General Public License as published by
 *    the Free Software Foundation, either version 3 of the License, or
 *    (at your option) any later version.
 *
 *    wiringPi is distributed in the hope that it will be useful,
 *    but WITHOUT ANY WARRANTY; without even the implied warranty of
 *    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *    GNU Lesser General Public License for more details.
 *
 *    You should have received a copy of the GNU Lesser General Public License
 *    along with wiringPi.  If not, see <http://www.gnu.org/licenses/>.
 ***********************************************************************
 */

#include <stdio.h>
#include <string.h>
#include <errno.h>

#include <wiringPi.h>
#include <wiringSerial.h>

int main ()
{
  int fd ;
  int count ;
  unsigned int nextTime ;

  if ((fd = serialOpen ("/dev/ttyACM0", 115200)) < 0)
  {
    fprintf (stderr, "Unable to open serial device: %s\n", strerror (errno)) ;
    return 1 ;
  }

  if (wiringPiSetup () == -1)
  {
    fprintf (stdout, "Unable to start wiringPi: %s\n", strerror (errno)) ;
    return 1 ;
  }

  nextTime = millis () + 300 ;

  for (count = 0 ; count < 256 ; )
  {
    if (millis () > nextTime)
    {
      printf ("\nOut: %3d: ", count) ;
      fflush (stdout) ;
      serialPutchar (fd, count) ;
      nextTime += 300 ;
      ++count ;
    }

    delay (3) ;

    while (serialDataAvail (fd))
    {
      printf (" -> %3d", serialGetchar (fd)) ;
      fflush (stdout) ;
    }
  }

  printf ("\n") ;
  return 0 ;
}`

hzeller, 2019-04-20:
ok, if you don't use GPIO, then wiringPi would be an option (However, in that case, I'd just use the regular Linux ways to configure the [serial interface via termios](https://en.wikibooks.org/wiki/Serial_Programming/termios), then simple open(); no need for wiringPi).

So if you _do_ want to use wiringPi: you're back to your original question making sure things link properly. I'd use the [description in the README how to create your own Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#integrating-in-your-own-application) as starting point, then just add the wiringPi library.

Toootless, 2019-04-20:
I will go back and reread that link when I tried to "add the wiringPi " to the marker file, it would start and go into the library and then exit and stops or it still does not link to the wiringPi files.

`make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix-master/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix-master/lib'
g++ display3.o -o display3 -L/usr/local/lib -L../lib -lrgbmatrix -lrt -lm -lpthread  -lstdc++ 
display3.o: In function `getData_convertData':
display3.c:(.text+0x2e4): undefined reference to `serialOpen'
display3.c:(.text+0x330): undefined reference to `wiringPiSetup'
display3.c:(.text+0x378): undefined reference to `millis'
display3.c:(.text+0x394): undefined reference to `millis'
display3.c:(.text+0x3d4): undefined reference to `serialPutchar'
display3.c:(.text+0x3f4): undefined reference to `delay'
display3.c:(.text+0x400): undefined reference to `serialGetchar'
display3.c:(.text+0x428): undefined reference to `serialDataAvail'
display3.c:(.text+0x454): undefined reference to `serialGetchar'
collect2: error: ld returned 1 exit status
Makefile:33: recipe for target 'display3' failed
make: *** [display3] Error 1
`

`CFLAGS=-Wall -O3 -g -Wextra -Wno-unused-parameter $(INCLUDE) -Winline -pipe
CXXFLAGS=$(CFLAGS)
INCLUDE	= -I/usr/local/include
LDFLAGS	= -L/usr/local/lib
LDLIBS    = -lwiringPi -lwiringPiDev -lpthread -lm -lcrypt -lrt

OBJECTS=demo-main.o minimal-example.o c-example.o text-example.o clock.o display3.o
BINARIES=demo minimal-example c-example text-example clock display3

# Where our library resides. You mostly only need to change the
# RGB_LIB_DISTRIBUTION, this is where the library is checked out.
RGB_LIB_DISTRIBUTION=..
RGB_INCDIR=$(RGB_LIB_DISTRIBUTION)/include
RGB_LIBDIR=$(RGB_LIB_DISTRIBUTION)/lib
RGB_LIBRARY_NAME=rgbmatrix
RGB_LIBRARY=$(RGB_LIBDIR)/lib$(RGB_LIBRARY_NAME).a
LDFLAGS+=-L$(RGB_LIBDIR) -l$(RGB_LIBRARY_NAME) -lrt -lm -lpthread 

all : $(BINARIES)

$(RGB_LIBRARY): FORCE
	$(MAKE) -C $(RGB_LIBDIR)

demo : demo-main.o $(RGB_LIBRARY)
	$(CXX) $< -o $@ $(LDFLAGS)

minimal-example : minimal-example.o
text-example: text-example.o
clock : clock.o

display3 : display3.o $(RGB_LIBRARY)
	$(CXX) $< -o $@ $(LDFLAGS) -lstdc++ 

# All the binaries that have the same name as the object file.q
% : %.o $(RGB_LIBRARY)
	$(CXX) $< -o $@ $(LDFLAGS)

# Since the C example uses the C++ library underneath, which depends on C++
# runtime stuff, you still have to also link -lstdc++
c-example : c-example.o $(RGB_LIBRARY)
	$(CC) $< -o $@ $(LDFLAGS) -lstdc++

%.o : %.cc
	$(CXX) -I$(RGB_INCDIR) $(CXXFLAGS) -c -o $@ $<

%.o : %.c
	$(CC) -I$(RGB_INCDIR) $(CFLAGS) -c -o $@ $<

clean:
	rm -f $(OBJECTS) $(BINARIES)

FORCE:
.PHONY: FORCE
`

hzeller, 2019-04-20:
You need to add `-lwiringPi -lwiringPiDev` to your `LDFLAGS` variable.

Toootless, 2019-04-20:
# Where our library resides. You mostly only need to change the
# RGB_LIB_DISTRIBUTION, this is where the library is checked out.
RGB_LIB_DISTRIBUTION=..
RGB_INCDIR=$(RGB_LIB_DISTRIBUTION)/include
RGB_LIBDIR=$(RGB_LIB_DISTRIBUTION)/lib
RGB_LIBRARY_NAME=rgbmatrix
RGB_LIBRARY=$(RGB_LIBDIR)/lib$(RGB_LIBRARY_NAME).a
LDFLAGS+=-L$(RGB_LIBDIR) -l$(RGB_LIBRARY_NAME) -lrt -lm -lpthread -lwiringPi -lwiringPiDev 

pi@raspberrypi:~/rpi-rgb-led-matrix-master/examples-api-use $ sudo make
make -C ../lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix-master/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix-master/lib'
g++ display3.o -o display3 -L/usr/local/lib -L../lib -lrgbmatrix -lrt -lm -lpthread -lwiringPi -lwiringPiDev  -lstdc++ 
pi@raspberrypi:~/rpi-rgb-led-matrix-master/examples-api-use $

hzeller, 2019-07-07:
Looks like you figured it out ?

closed 2019-07-07 by hzeller