hzeller/rpi-rgb-led-matrix #801: Outdated warning for adafruit installer?

marcmerlin, 2019-04-20:
README.md says
"Switch the Pinout
The Adafruit HAT/Bonnet uses this library but a modified pinout to support other features on the HAT. So they forked this library and modified the pinout there. However, that fork is ancient, so I strongly suggest to use this original library instead. You can choose the Adafruit pinout with a command line flag."

Their current installer gets
GITUSER=https://github.com/hzeller
REPO=rpi-rgb-led-matrix
COMMIT=814b79b5696d32dd1140304b41a1ec0068bb271a

Looks like it grab https://github.com/hzeller/rpi-rgb-led-matrix/commit/814b79b5696d32dd1140304b41a1ec0068bb271a
which is not the latested, but fewer than 10 revs behind.

Worth updating the doc?

hzeller, 2019-04-20:
Do you have a link to their page where they use the new library ?
In the past, they used the library, but then modify the Makefile to set their HAT to the default.

marcmerlin, 2019-04-21:
Yes, they tweak the build to default to their hat/bonnet
https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi?view=all
mentions
```
curl https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rgb-matrix.sh >rgb-matrix.sh
sudo bash rgb-matrix.sh
```
which does
```
if [ $QUALITY_MOD -eq 1 ]; then
        # Build then install for Python 2.7...
        make build-python HARDWARE_DESC=adafruit-hat-pwm
        make install-python HARDWARE_DESC=adafruit-hat-pwm
```

marcmerlin, 2019-04-21:
But they've stopped shipping a very outdated version 
```
Earlier versions of this guide used our own fork of this library. That’s deprecated now, but still available if you have existing code built atop it. Otherwise, use this installer script and latest code.
```

commit a20d39ce (2019-07-07): Adafruit uses this library directly these days.

Fixes #801
closed 2019-07-07 by hzeller