hzeller/rpi-rgb-led-matrix #802: Allow an Adafruit::NeoMatrix style panel arrangement and give examples of recommended wirings

marcmerlin, 2019-04-21:
After reading the docs, it seems that to make a rectangular or square array with 4 panels or more, the only supported layout is --led-pixel-mapper="U-mapper"

Beyond that, it seems that I have to write a 
```
bool RGBMatrix::ApplyPixelMapper(const PixelMapper *mapper);
```
It would be awesome to support some syntax similar to what adafruit allows as explained in https://learn.adafruit.com/adafruit-neopixel-uberguide/neomatrix-library
[image]

For instance, my current panel layout is

In P1 Out     In P4 Out
In P2 Out     In P5 Out
In P3 Out     In P6 Out

NEO TILE_TOP + NEO_TILE_LEFT + NEO_TILE_COLUMNS + NEO_TILE_PROGRESSIVE would define this layout

More generally with RGBPanels, it would make wiring short if every other panel could be upside down. The shortest wiring would be:
In P1 Out     In P6 Out
Out P2 In     Out P5 In
In P3 Out     In P4 Out
but this requires mapping P2 and P5 with pixels upside down, which sucks more code-wise.

Having some recommendations of best layouts wiring vs code-wise, would be helpful

marcmerlin, 2019-04-22:
Feel free to take the XY() logic from NeoMatrix
https://github.com/marcmerlin/FastLED_NeoMatrix/blob/e858b08b9b54f27a674cbd9a6b29a912e4cd7f84/FastLED_NeoMatrix.cpp#L124

hzeller, 2019-07-07:
Ideally, I'd like to have something that can be configured with a single commandline, similar to parameters to the U-mapper. Pull requests welcome.

marcmerlin, 2020-03-16:
This is close enough to https://github.com/hzeller/rpi-rgb-led-matrix/issues/1003 that I think it's fair to make them dupes.

closed 2020-03-16 by marcmerlin