hzeller/rpi-rgb-led-matrix #803: Hub75 1:4 P10 Outdoor panel (16x32) using Adafruit Bonnet

gjs56, 2019-04-23:
Hi,
I'm trying to get the above LED matrix working with my RasPi 3 Model A Plus.  Despite running examples with different options on multiplexing, row-addr, slowdown and others I have had no success at all - not even a flicker from a single LED. I feel as though I'm missing something fundamental since all other issues that I've looked at using similar configurations have at least had some initial output from the panel.
There is power to the bonnet (green light) which I believe is connected correctly to the panel.  There is a ribbon cable (provided by the supplier of the panel) between the bonnet and the panel. The pi is powered via the bonnet and isn't connected to anything else (monitor, keyboard etc) and I'm using Putty to submit commands to the pi from my laptop.
When running a sample program I get a message indicating that gpio mapping is via adafruit-hat and the program exits after reaching timeout.
Unfortunately I am an absolute beginner when it comes to the pi and have practically no knowledge of electronics.  
If anyone can point me in the right direction or just give me some checks to make I'd be most grateful.

davemaster, 2019-04-23:
pictures???

gjs56, 2019-04-24:
Thanks for rapid response. Pictures as requested.  Forgot to mention that I have also disabled the audio parameter in the pi config as recommended.

[image]
[image]
[image]

ChristopherBadenhorst, 2019-04-29:
Have you soldered the pins to the bonnet?

gjs56, 2019-04-30:
From Adafruit's description of the bonnet, no soldering is required. [](https://www.adafruit.com/product/3211)

ChristopherBadenhorst, 2019-05-04:
Weird, using the same panels with the passive-3 (and wired pinout like in the 'readme') and that works. 
Have you tried wiring up directly?

gjs56, 2019-05-04:
I have previously tried using individual wires but couldn't get anything working.  I was on the verge of giving up when I saw Adafruit's bonnet and thought I'd give it a go.  It feels like there's a secret switch somewhere or some additional configuration that I need to change.  Is there a way to ask the Pi what hardware it recognises (akin to plugging a camera into a Windows PC)?

ChristopherBadenhorst, 2019-05-04:
Got the same panels and that works fine. Just bought a Ada Hat and will try that too. Will try and keep track of the steps I do.
Ever thought of the panel being broken?

gjs56, 2019-05-04:
I have considered there being a problem with the panel but I do have two and neither is delivering anything.  Many thanks for trying to help with this.

davemaster, 2019-05-05:
try using the following:
--led-gpio-mapping=regular (and connect directly with cables)
the you need to try with
--led-rows=16
--led-cols=32
--led-chain=1 or 2 or 3 (depends on the number of panels in chain -12 max according readme-)
--led-parallel=1
--led-multiplexing=9 (try another if it doesn't works)

davemaster, 2019-05-05:
Example

(while :; do date +%T ; sleep 0.5 ; done) | sudo ./text-example -f ../fonts/8x13B.bdf -y2 -C255,0,0 --led-cols=32 --led-rows=16 --led-chain=2 --led-multiplexing=8 --led-pwm-lsb-nanoseconds 250 --led-brightness=20