hzeller/rpi-rgb-led-matrix #804: Image distortion

ChristopherBadenhorst, 2019-04-29:
Hi,

Using the library on a combination with 2x3 panels, with a passive-3 I soldered. The hat works fine with a 32x32 p0.5 and the images are correct, but really distorted.

https://github.com/login?return_to=%2Fhzeller%2Frpi-rgb-led-matrix%2Fissues

I already tried the gpio slowdown, and this is the best result. 

Wil it work better with the active-3?

Regards,

Chris

ChristopherBadenhorst, 2019-05-01:
Found the solution in the other parameters. 
I only need the '--led-pwm-dither-bits' in python. Anyone who can help?

ChristopherBadenhorst, 2019-05-04:
Got myself an Adafruit RGB + RTC. Works like a charm.

hzeller, 2019-07-07:
PWM dither bits have been added in 4c9a4036f44f173b783a0ba701138605f8b7d95c

closed 2019-07-07 by hzeller