hzeller/rpi-rgb-led-matrix #805: 128x64

PHI34Halladay, 2019-04-29:


marcmerlin, 2019-05-01:
yeah, a picture will likely help

PHI34Halladay, 2019-05-02:
Hey Marc,

thanks for your answer. Here is a picture from the matrix
[image]

[image]

trevor-viljoen, 2019-05-02:
What parameters are you passing to the script? You would need something like ```—led-rows=64 —led-cols=32 —led-chain=2 —led-parallel=2```

PHI34Halladay, 2019-05-03:
Yes, I tried the exact same LED flags, but it says that my Pi is not able to habe two LEDs parallel. That's why I tried it like that:
--led-rows=64 --led-cols=128 --led-chains=2

I'm using the newest Raspberry Pi btw.

Beersatron, 2019-05-03:
I can't quite see in the picture, but does your HAT have two outputs or just the one?

In other words, what way are the panels connected?

Pi -> HAT -> Panel1 -> Panel2 - Panel3 - Panel4
or
Pi -> HAT -> Panel1 -> Panel2
            \_> Panel3 -> Panel 4

PHI34Halladay, 2019-05-03:
It has only one output.

trevor-viljoen, 2019-05-03:
Are you able to get one panel to setup correctly? ```--led-rows=32 --led-cols=64``` You might try disconnecting them and then making sure a single panel works, then add another: ``` --led-rows=32 --led-cols=64 --led-chain=2```

PHI34Halladay, 2019-05-03:
1 panel works perfectly and I also don't have any problem using two panels.

Beersatron, 2019-05-03:
--led-pixel-mapper="U-mapper" --led-rows=32 --led-cols=64 --led-chain=4

trevor-viljoen, 2019-05-03:
@PHI34Halladay, I have a feeling this may be mlb-led-scoreboard related. Can you run this command and post the output? ```sudo python main.py --led-rows=32 --led-cols=64 --led-pixel-mapper=U-mapper --led-chain=4```

I don't think mlb-led-scoreboard is able to determine the layout. If that is the case, let's close this issue and re-open one for that project.

PHI34Halladay, 2019-05-04:
> @PHI34Halladay, I have a feeling this may be mlb-led-scoreboard related. Can you run this command and post the output? `sudo python main.py --led-rows=32 --led-cols=64 --led-pixel-mapper=U-mapper [...]

I ran that command and nothing happened. The script loaded normally but the LEDs keep dark.

I've seen pictures from the mlb-led-scoreboard where guys are using 128x64 on 4 LEDs so it should work somehow.

PHI34Halladay, 2019-05-04:
Guys, I got it done...
I played around with the LED flags for a little while and endet up with that command:

sudo python main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping="adafruit-hat" --led-pixel-mapper=U-mapper

Thanks for all the help and especially the U-mapper info :)

closed 2019-05-04 by PHI34Halladay

angelogoncalve, 2019-05-04:
Yes you are correct:

bonnet hat - flag: --led-gpio-mapping="adafruit-hat"

one output that link 4 panels in serie - flag: --led-rows=32 --led-cols=64 --led-chain=4

you have 2 rows of panels that the message are equal for both - flag: --led-pixel-mapper=U-mapper